NVIDIA/warp #368: [QUESTION] wp.array of struct: Unsupported NumPy data type 'object'

ucacaxm, 2024-11-26:
Hi all,

Many thanks for the job around Warp !

I would like to create an wp.array of struct.
I create an `numpy.array` of the struct. The type seems ok, but
when I try to create the wp.array from the numpy array  causes an error:
```
Unsupported NumPy data type 'object'.
```

```
import numpy as np
import warp as wp
from warp.types import vector

@wp.struct
class JointCfg:
    pid: wp.int32
    l2p: wp.transformf

    def __str__(self):
        return f"JointState(pid={self.pid}, l2p={self.l2p}"

pid = [ -1, 0, 1, 1, 1, 3, 4 ]
nJoint = len(pid)
skeCfgNP = []
for i in range( nJoint):
    j = JointCfg()
    j.pid = wp.int32(pid[i])
    j.l2p = wp.transform( wp.vec3(i,0,0), wp.quat_identity() )
    skeCfgNP.append( j )
skeCfgNP = np.array( skeCfgNP )

print( type(skeCfgNP) )
# => <class 'numpy.ndarray'>
print( type(skeCfgNP[0]) )
# => <class 'warp.codegen.Struct.__call__.<locals>.NewStructInstance'>

for i in range( nJoint):
    print( skeCfgNP[i] )
# => OK it prints the JointState

skeCfg = wp.from_numpy( skeCfgNP )
```
The error is from the last line.

I tried to remove the `@wp.struct` but with the same error.
Any help please ?

Thanks

shi-eric, 2024-11-28:
Is the step of creating a NumPy array important, or can you avoid this issue by creating a `wp.array` from the `skeCfgNP` list?

ucacaxm, 2024-11-28:
Yes this is what i did.
I created an empty wp.array and fill it with a kernel.
```
skeCfg = wp.zeros( dtype=JointCfg, shape=nJoint )
wp.launch(kernel=initSkeletonConfig, dim=1, inputs=[skeCfg, wp_pid, wp_tr] )
```

By the way, just one question: can I reshape an array in a kernel?
If I add an array to the struct 
```
@wp.struct
class JointCfg:
    pid: wp.int32
    l2p: wp.transformf
    children = wp.array( dtype=wp.int32) 
```

Can I change the size of `children` in the kernel ?
I think no.

Thanks

shi-eric, 2024-11-28:
No, you can't reshape an array in a kernel.

For completeness, I think you could have also avoided the kernel launch in the following way:

```python
import warp as wp

@wp.struct
class JointCfg:
    pid: wp.int32
    l2p: wp.transformf

    def __str__(self):
        return f"JointState(pid={self.pid}, l2p={self.l2p}"

pid = [-1, 0, 1, 1, 1, 3, 4]
nJoint = len(pid)
skeCfgList = []
for i in range(nJoint):
    j = JointCfg()
    j.pid = wp.int32(pid[i])
    j.l2p = wp.transform(wp.vec3(i, 0, 0), wp.quat_identity())
    skeCfgList.append(j)
skeCfgWP = wp.array(skeCfgList, dtype=JointCfg)
```

closed 2024-11-28 by ucacaxm