NVIDIA/warp #372: [BUG] Warp Sim examples segfault on Mac ARM CPU

eric-heiden, 2024-11-29:
### Bug Description

Running `example_cartpole.py` on Mac with Apple Silicon chips (e.g. M1 MacBook Pro) fails, need to investigate further:

```
Warp 1.4.2 initialized:
   CUDA not enabled in this build
   Devices:
     "cpu"      : "arm"
   Kernel cache:
     /Users/Myself/Library/Caches/warp/1.4.2
Module warp.sim.collide f6c82b4 load on device 'cpu' took 7.96 ms  (cached)
Module warp.sim.articulation aada68e load on device 'cpu' took 1506.18 ms  (compiled)
Segmentation fault: 11
```

wgu93, 2024-12-06:
Hi everyone, I encounter the same issue on M3 MacBook Pro in a virtual environment with Python=3.10.

Run:
```
python example_bounce.py 
```

Results:
```
Warp 1.5.0 initialized:
   CUDA not enabled in this build
   Devices:
     "cpu"      : "arm"
   Kernel cache:
     /Users/wgu/Library/Caches/warp/1.5.0
Module warp.sim.collide f3f8e69 load on device 'cpu' took 1.78 ms  (cached)
zsh: segmentation fault  python example_bounce.py
```

shi-eric, 2025-02-25:
This issue seems related to #552. After adding some debug commands into `example_cartpole.py`:

```
import faulthandler

faulthandler.enable()

wp.config.mode = "debug"
```

I can get:

```
Warp 1.6.0 initialized:
   CUDA not enabled in this build
   Devices:
     "cpu"      : "arm"
   Kernel cache:
     /Users/ershi/Library/Caches/warp/1.6.0
Module warp.sim.collide 0854a6f load on device 'cpu' took 0.54 ms  (cached)
Module warp.sim.articulation c03ef51 load on device 'cpu' took 1.28 ms  (cached)
Fatal Python error: Segmentation fault

Current thread 0x00000001ed358840 (most recent call first):
  File "/Users/ershi/code-projects/warp/warp/context.py", line 5683 in launch
  File "/Users/ershi/code-projects/warp/warp/sim/articulation.py", line 489 in eval_fk
  File "/Users/ershi/code-projects/warp/warp/examples/sim/example_cartpole.py", line 87 in __init__
  File "/Users/ershi/code-projects/warp/warp/examples/sim/example_cartpole.py", line 135 in <module>

Extension modules: numpy._core._multiarray_umath, numpy.linalg._umath_linalg (total: 2)
Segmentation fault: 11
```

But this is only with Python installations I installed from `uv`. The ones I installed from Brew and Miniforge seem to work fine...

vyeevani, 2025-05-24:
running into this issue as well

groves152, 2025-06-03:
This is still an issue on Python 3.11.5 and Warp 1.7.2 on M3 Macbook Pro

Warp 1.7.2 initialized:
   CUDA not enabled in this build
   Devices:
     "cpu"      : "arm"
   Kernel cache:
     /Users/josh/Library/Caches/warp/1.7.2
Module lib.pulse f20debb load on device 'cpu' took 6.55 ms  (cached)
zsh: segmentation fault  /Applications/miniconda/miniconda3/envs/nvidia_warp/bin/python

joshuayanggithub, 2025-06-20:
Have any of you been able to resolve this issue? @groves152 @vyeevani @shi-eric

c0d1f1ed, 2025-07-25:
Thanks everyone for reporting this! It has the same root cause as #356