NVIDIA/warp #383: [BUG] parse_mjcf ?

ucacaxm, 2024-12-06:
### Bug Description

Hi,

I have trouble with `parse_mjcf`
I load the same xml file with Mujoco and with Warp. The Warp version is wrong.
I attache two images: the first one with Mujoco and the second with Warp. 

[image]
[image]

I have the same problem with the `ant.xml` used in the regression test.
It looks like a transform issue.
I do not compute any physics: just load, finalize the builder and display. 
The 'y' axis is well set (Y for ant and Z for my file).
I use the `SimRendererOpenGL`.
Could it be the Renderer?

Otherwise I will have a look at the `parse_mjcf` funtion

Thanks,

### System Information

Windows
anaconda + python 3.10.15
Warp 1.5

eric-heiden, 2024-12-09:
Hi @ucacaxm,

Thanks for reporting the issue! Would you be able to share the MJCF file so we can use it for debugging?

ucacaxm, 2024-12-09:
Maybe it comes from my code, since the bug exists also with the ant.xml file.
I put my code further.

```
<mujoco>
  <asset>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1="0.2 0.2 0.2" rgb2="0.8 0.8 0.8"/>
    <material name="grid" texture="grid" texrepeat="10 10" reflectance="0.0"/>
  </asset>

  <worldbody>
    <geom name="floor" size="100 100 .05" type="plane" material="grid" condim="3" pos="0 0 0" euler="0 0 0" friction="0.1 0.1 0.1"/>

    <!-- Upper Leg -->
    <body name="thigh" pos="0 0 2">
        <joint name="hip" type="free"/>
        <geom name="hip_sph" type="sphere" size="0.1" rgba="1 0 0 1"/>
        <geom name="hip_geom" type="box" pos="0 0 -0.25" size="0.06 0.06 0.16" rgba="1 0 0 1"/>
    
        <!-- Lower Leg -->
        <body name="shin" pos="0 0 -0.5">
            <joint name="knee" type="hinge" axis="0 1 0" range="0 90"/>
            <geom name="knee_sph" type="sphere" size="0.1" rgba="0 1 0 1"/>
            <geom name="knee_geom" type="box" pos="0 0 -0.25" size="0.05 0.05 0.15" rgba="0 1 0 1"/>
            
            <!-- Foot -->
            <body name="foot" pos="0 0 -0.5">
                <joint name="ankle_x" type="hinge" axis="1 0 0" range="-45 45"/>
                <joint name="ankle_z" type="hinge" axis="0 0 1" range="-45 45"/>
                <geom name="foot_sph" type="sphere" size="0.1" rgba="0 0 1 1"/>
                <geom name="foot_geom" type="box" pos="0.12 0 -0.1" size="0.25 0.08 0.04" rgba="0 0 1 1"/>
            </body>
        </body>
    </body>

  </worldbody>

  <actuator>
    <motor name="knee"          gear="140"  joint="knee"/>
    <motor name="ankle_x"       gear="40"  joint="ankle_x"/>
    <motor name="ankle_z"       gear="40"  joint="ankle_z"/>
  </actuator>

</mujoco>
```

My code :

```

import math
import os
import numpy as np
import warp as wp
import warp.examples
import warp.sim
import warp.sim.render
#from mjcf import custom_parse_mjcf

class Example:
    def __init__(self):
        builder = wp.sim.ModelBuilder()
        wp.sim.parse_mjcf(
        #custom_parse_mjcf(
            "data/mujoco/oneLeg.xml",
            builder,
            armature=1.0,
            stiffness=0,
            damping=1.0,
            contact_ke=1.0e4,
            contact_kd=1.0e2,
            contact_kf=1.0e2,
            contact_mu=0.75,
            limit_ke=1.0e3,
            limit_kd=1.0e1,
            up_axis="Z",
        )

        self.model = builder.finalize()
        self.model.ground = True

        self.renderer = wp.sim.render.SimRendererOpenGL(self.model, "") #stage_path)

        self.state_0 = self.model.state()

    def render(self):
        with wp.ScopedTimer("render", print=False):
            self.renderer.begin_frame( 0.0 )
            self.renderer.render(self.state_0)
            self.renderer.end_frame()

    def run(self, device=None):
        with wp.ScopedDevice(device):
            while self.renderer.is_running():
                self.render()

if __name__ == "__main__":
    viewer = Example()
    viewer.run()
```

ucacaxm, 2024-12-09:
Sorry I have to call `eval_fk` to run one pass of forward kinematics ...
Like this:

```

import math
import os
import numpy as np
import warp as wp
import warp.examples
import warp.sim
import warp.sim.render
#from mjcf import custom_parse_mjcf

class Example:
    def __init__(self):
        builder = wp.sim.ModelBuilder()

        wp.sim.parse_mjcf(
        #custom_parse_mjcf(
            "data/mujoco/oneLeg.xml",
            builder,
            armature=1.0,
            stiffness=0,
            damping=1.0,
            contact_ke=1.0e4,
            contact_kd=1.0e2,
            contact_kf=1.0e2,
            contact_mu=0.75,
            limit_ke=1.0e3,
            limit_kd=1.0e1,
            up_axis="Z",
        )

        self.model = builder.finalize()
        self.model.ground = True

        self.renderer = wp.sim.render.SimRendererOpenGL(self.model, "") #stage_path)

        self.state_0 = self.model.state()
        **wp.sim.eval_fk(self.model, self.model.joint_q, self.model.joint_qd, None, self.state_0)**

    def render(self):
        with wp.ScopedTimer("render", print=False):
            self.renderer.begin_frame( 0.0 )
            self.renderer.render(self.state_0)
            self.renderer.end_frame()

    def run(self, device=None):
        with wp.ScopedDevice(device):
            while self.renderer.is_running():
                self.render()

if __name__ == "__main__":
    viewer = Example()
    viewer.run()
```

closed 2024-12-09 by ucacaxm