NVIDIA/warp #395: [QUESTION] <title> If anyone will get a Nan graident when try to learn the soft body material like k_mu?

zhenqi72, 2024-12-13:
Hi, I am trying to learn the soft body material, according to the difference between a stand soft body and a learning soft body. The scenario is that a capsule will push a soft body. But I tried lots of ways. Both of them will make the gradient to "nan".

This is my code. 
```
  import math
  import sys
  import os
  sys.path.append(os.path.dirname(os.path.dirname(os.path.abspath(__file__))))
  import warp as wp
  import warp.sim
  import warp.sim.render
  import numpy as np
  from warp_test_folder.adam import Adam
  
  import torch   
  
  @wp.kernel
  def loss_calculate(
      pos: wp.array2d(dtype=wp.vec3),
      target_pos: wp.array2d(dtype=wp.vec3),
      loss_element: wp.array2d(dtype=float),
  ):
      i,j = wp.tid()
      loss_element[i][j] = wp.length(target_pos[i][j] - pos[i][j])
  
  @wp.kernel
  def loss_sum(loss_element: wp.array2d(dtype=float), result: wp.array(dtype=float)):
  
      i,j = wp.tid()
      wp.atomic_add(result, 0, loss_element[i][j])
      
  @wp.kernel
  def save_state(particle_q: wp.array(dtype=wp.vec3), write_index: int, last_traj: wp.array2d(dtype=wp.vec3f)):
      i = wp.tid()
      last_traj[write_index][i] = particle_q[i]
      
  @wp.kernel
  def minus(grad: wp.array(dtype=float, ndim=2), orig: wp.array(dtype=float, ndim=2), mid: wp.array(dtype=float, ndim=2)):
      tid = wp.tid()
      mid[tid][0] = orig[tid][0] - grad[tid,0]
      orig[tid][0] = mid[tid][0]
  
  class Example:
      def __init__(self, stage_path="soft_body.usd", num_frames=30):
          self.sim_substeps = 10
          self.num_frames = num_frames
          self.fps = 60
          sim_duration = self.num_frames / self.fps
          self.frame_dt = 1.0 / self.fps
          self.sim_dt = self.frame_dt / self.sim_substeps
          self.sim_time = 0.0
          self.lift_speed = 2.5 / sim_duration * 2.0  # from Smith et al.
          self.rot_speed = math.pi / sim_duration
          self.tape = wp.Tape()
          self.target = torch.load("learn_grad_pos.pt")
          self.ref_traj = wp.from_torch(self.target,requires_grad=True,dtype=wp.vec3f).to('cuda:3')
          self.last_traj = wp.empty_like(self.ref_traj)
          self.loss_element = wp.zeros(shape=(self.ref_traj.shape[0:2]), dtype = float, device='cuda:3', requires_grad=True)
          self.result = wp.zeros(shape=(1,), dtype = float, device='cuda:3', requires_grad=True)
    
          capsule_builder = wp.sim.ModelBuilder() 
          b = capsule_builder.add_body(m=10.0,name="capsule0")#(1.0,1.55, 5.0)
          a = capsule_builder.add_body(m=10.0,name="connect body") # if you has the joint, body position will be decided by joint xform
          capsule_builder.add_shape_capsule(body = b,radius=0.5, half_height=1.0, density=100.0)
  
          capsule_builder.add_joint_prismatic(
              parent=-1,
              child=a,
              axis=(1,0,0),
              target_ke = 400,
[... 159 lines cut ...]
      parser.add_argument("--device", type=str, default='cuda:3', help="Override the default Warp device.")
      parser.add_argument(
          "--stage_path",
          type=lambda x: None if x == "None" else str(x),
          default="soft_cap_grad_learn_copy.usd",
          help="Path to the output USD file."
      )
      parser.add_argument("--num_frames", type=int, default=30, help="Total number of frames.") # 900
      parser.add_argument("--train_iters", type=int, default=150, help="Total number of training iterations.")
      args = parser.parse_known_args()[0]
  
      with wp.ScopedDevice(args.device):
          example = Example(stage_path=args.stage_path, num_frames=args.num_frames)
  
          for iteration in range(args.train_iters):
              print(f"Iteration {iteration}")
              example.step(iteration)
              #example.render()
              example.states = []
              for _ in range(example.num_frames + 1):
                  example.states.append(example.model.state())
  
          if example.renderer:
              example.renderer.save()
```

Jianghanxiao, 2024-12-13:
Some suggestions based on my experience, you'd better make sure all intermediate variables are saved somewhere. Especially in your substep loop, you don't save the next state explicitly. After you modifying the code, you can check the gradient after backward to see which part is broken, which makes life easier

MontaEllis, 2025-01-07:
Have you solved the problem? I met the same problem.

zhenqi72, 2025-01-07:
> Have you solved the problem? I met the same problem.

No， I am still struggling with it.

MontaEllis, 2025-01-08:
In my code, the gradient is zero. It's strange.

[image]

zhenqi72, 2025-01-08:
> In my code, the gradient is zero. It's strange.
> 
> [image]

Maybe check your loss and the state your stored during the forward. My problem right now is the gradient keep going up and turn to NaN

MontaEllis, 2025-01-08:
Now my issue is similar to yours.

[image]

closed 2025-01-08 by shi-eric