NVIDIA/warp #403: [BUG] tile_arange() has issues with negative numbers

shi-eric, 2024-12-19:
### Bug Description

Example:

```python
import warp as wp

@wp.kernel
def compute():
    a = wp.tile_arange(-10, 10, dtype=int)

    print(a)

wp.launch(compute, dim=[16], inputs=[])
```

This code results in the following issue:

```
  File "/home/eshi/code-projects/warp/warp/builtins.py", line 1856, in tile_arange_value_func
    raise RuntimeError("wp.tile_arange() arguments must be compile time constants")
RuntimeError: Error while parsing function "compute" at /home/eshi/code-projects/warp/ershi-tests/test_tile_min_docstring.py:6:
    a = wp.tile_arange(-10, 10, dtype=int)
;wp.tile_arange() arguments must be compile time constants
```

(Note also that the `RuntimeError` raised in `tile_arange_value_func()` is not very helpful here).

Similarly, using a negative step doesn't work:

```python
import warp as wp

@wp.kernel
def compute():
    a = wp.tile_arange(10, 0, -1, dtype=int)

    print(a)

wp.launch(compute, dim=[16], inputs=[])
```

Output:

```
  File "/home/eshi/code-projects/warp/warp/builtins.py", line 1856, in tile_arange_value_func
    raise RuntimeError("wp.tile_arange() arguments must be compile time constants")
RuntimeError: Error while parsing function "compute" at /home/eshi/code-projects/warp/ershi-tests/test_tile_min_docstring.py:6:
    a = wp.tile_arange(10, 0, -1, dtype=int)
;wp.tile_arange() arguments must be compile time constants
```

daedalus5, 2025-01-03:
Looking into this a bit, `-1` is not a compile-time constant (which is required for this builtin to be able to choose the right tile template). It's a unary operator (`-`) and the constant 1. So our AST parser breaks this into a separate runtime statement. This is a good design for differentiability, but also a nuisance in this case, because the "fix" is to write this line instead like:
`wp.tile_arange(wp.static(-10), 10)`

which is not the friendliest syntax. I've been trying to see if we can automate the static interpretation of these args, but it remains to be seen.

shi-eric, 2025-01-08:
Fixed by af24fa8a4ee8595ef6ac93c91186dec30f508c13