NVIDIA/warp #405: [QUESTION]  `self.model.enable_tri_collisions = True` makes `particle_q = state.particle_q.numpy()` super slow

ywu110, 2024-12-21:
## Issue Overview:

I am working on a cloth simulation using the [example provided here](https://github.com/NVIDIA/warp/blob/5b7a93415784350599b91ad28c9aba89e12b624b/warp/examples/sim/example_cloth.py#L9). In addition to handling collisions between the cloth and the Bunny model, I aim to incorporate cloth self-collision into the simulation.

## Steps Taken:

1. **Enabled Self-Collision**: I added the following line to the example code to enable cloth self-collision:
```python
self.model.enable_tri_collision = True
```

2. **Observed Issues**:
- **Persistent Penetration**: Despite enabling self-collision, the simulation still exhibits significant penetration within the cloth itself.

- **Performance Degradation**: Enabling `self.model.enable_tri_collision` drastically increased the rendering time from approximately 2ms to 800ms. Using `wp.ScopedTimer`, I traced the slowdown to the following line in `warp/sim/render.py`:

```python
particle_q = state.particle_q.numpy()
```
[Link to the specific line](https://github.com/NVIDIA/warp/blob/75544dfa34d46262e05900c13fa66d91f35e63dd/warp/sim/render.py#L318)

## Questions:

1. **Self-Collision Effectiveness**:

- Is adding `self.model.enable_tri_collision = True` sufficient to enable effective cloth self-collision? 

- If not, what additional steps or configurations are necessary to achieve reliable self-collision handling?

2. **Performance Impact**: 

- Why does enabling `self.model.enable_tri_collision` cause such a significant increase in rendering time, especially when the shape of `particle_q `remains unchanged?
- Are there recommended optimizations or alternative approaches to mitigate this performance hit while maintaining accurate self-collision?

## Additional Information:

Warp Version: Both V1.3.0 and V1.5.0
Hardware: GPU 3090 CUDA 12.2 Driver Version: 535.183.01
Operating System: Ubuntu 22.04
Python Version: Python 3.10.13

## Thank you in advance!! 
I am keen to resolve these issues to advance my simulation work. Any guidance or suggestions you can provide would be greatly appreciated.
Thank you once again for developing and maintaining such an amazing project!

shi-eric, 2024-12-30:
I can't speak about the `enable_tri_collision` flag, but I did want to comment on the performance analysis and caution about `wp.ScopedTimer`. Since the CUDA API asynchronous, this can lead to misleading results when combined with traditional timing libraries in Python. Our profiling docs goes into this: https://nvidia.github.io/warp/profiling.html

In your specific case, the hit in performance isn't coming from `state.particle_q.numpy()`. Instead, it seems that the `eval_triangles_contact` kernel is quite expensive (https://github.com/NVIDIA/warp/blob/5b7a93415784350599b91ad28c9aba89e12b624b/warp/sim/integrator_euler.py#L259-L316), adding a kernel that takes ~28 ms onto a substep that used to have ~0.02 ms of GPU operations!

To make these conclusions, I needed to modify the example to:

- Have `synchronize=True` to get more accurate timings for the `step` and `render` timers.
- Turn off CUDA graphs to get increased granularity in the timeline when profiled with Nsight Systems. Not using CUDA graphs does have a negative impact on performance, but we already know what the overall performance hit of using `self.model.enable_tri_collisions = True` was.

In this image, you can see that the GPU activity is dominated by these `eval_triangles_contact` calls:
[image]

AnkaChan, 2024-12-30:
Hi 冰糖葫芦，
What integrator are you using? I assume you are using the Euler integrator, since that is the default option.
Enabling `enable_tri_collision` adds point-triangle collision response, which by itself is not very stable. Furthermore, the collision detection uses a brute-force search, leading to an O(𝑁^2)complexity.

That is the why the simulation becomes slower after turning on `enable_tri_collision`. The reason why rendering time goes up is because the `simulate()` function only launches the simulation kernel without waiting for it to finish. However, rendering requires synchronization, and most of the time of the rendering step is waiting for the launched kernels to finish.

To be honest, achieving stable collision responses in an explicit integrator is very challenging the least to say. 
However, currently only the Euler integrator, which is an explicit integrator, supports self-collision. XPBD and VBD do not. But the good news is that we are planing to release a new version in January that introduces a robust self-collision handler for the VBD integrator, which we believe is the best solution for this issue. The new integrator will provide penetration-free guarantee for robust elasticity simulation.

AnkaChan, 2025-03-03:
Hi 冰糖葫芦，
I’m delighted to announce that we’ve released the self-collision feature in the VBDIntegrator! Simply upgrade to version 1.6.0 and initialize VBDIntegrator with handle_self_contact=handle_self_contact. This will automatically handle cloth self-collision while ensuring a penetration-free simulation.

We’ve also provided an example of self-collision handling in [warp/examples/sim/example_cloth_self_contact.py](https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_cloth_self_contact.py), please refer to it for more detailed instructions.

closed 2025-03-03 by AnkaChan