NVIDIA/warp #406: [Error]Warp CUDA error 999: unknown error

Valerianding, 2024-12-25:
Hi, I'am using the example below:
```
# Copyright (c) 2022 NVIDIA CORPORATION.  All rights reserved.
# NVIDIA CORPORATION and its licensors retain all intellectual property
# and proprietary rights in and to this software, related documentation
# and any modifications thereto.  Any use, reproduction, disclosure or
# distribution of this software and related documentation without an express
# license agreement from NVIDIA CORPORATION is strictly prohibited.

###########################################################################
# OpenGL renderer example
#
# Demonstrates how to set up tiled rendering and retrieves the pixels from
# OpenGLRenderer as a Warp array while keeping all memory on the GPU.
#
###########################################################################

import numpy as np

import warp as wp
import warp.render

wp.init()

class Example:
    def __init__(self, num_tiles=4, custom_tile_arrangement=False):
        # import pdb
        # pdb.set_trace()
        self.renderer = wp.render.OpenGLRenderer(vsync=False)
        instance_ids = []

        if custom_tile_arrangement:
            positions = []
            sizes = []
        else:
            positions = None
            sizes = None

        if num_tiles > 1:
            # set up instances to hide one of the capsules in each tile
            for i in range(num_tiles):
                instances = [j for j in np.arange(13) if j != i + 2]
                instance_ids.append(instances)
                if custom_tile_arrangement:
                    angle = np.pi * 2.0 / num_tiles * i
                    positions.append((int(np.cos(angle) * 150 + 250), int(np.sin(angle) * 150 + 250)))
                    sizes.append((150, 150))
            self.renderer.setup_tiled_rendering(instance_ids, tile_positions=positions, tile_sizes=sizes)

        self.renderer.render_ground()

    def render(self):
        time = self.renderer.clock_time
        self.renderer.begin_frame(time)
        for i in range(10):
            self.renderer.render_capsule(
                f"capsule_{i}",
                [i - 5.0, np.sin(time + i * 0.2), -3.0],
                [0.0, 0.0, 0.0, 1.0],
                radius=0.5,
                half_height=0.8,
            )
        self.renderer.render_cylinder(
            "cylinder",
            [3.2, 1.0, np.sin(time + 0.5)],
            np.array(wp.quat_from_axis_angle(wp.vec3(1.0, 0.0, 0.0), wp.sin(time + 0.5))),
            radius=0.5,
            half_height=0.8,
        )
        self.renderer.render_cone(
            "cone",
            [-1.2, 1.0, 0.0],
            np.array(wp.quat_from_axis_angle(wp.vec3(0.707, 0.707, 0.0), time)),
            radius=0.5,
            half_height=0.8,
        )
        self.renderer.end_frame()

if __name__ == "__main__":
    import argparse
    import distutils.util

    parser = argparse.ArgumentParser(formatter_class=argparse.ArgumentDefaultsHelpFormatter)
    parser.add_argument("--device", type=str, default=None, help="Override the default Warp device.")
    parser.add_argument("--num_tiles", type=int, default=4, help="Number of viewports to render in a single frame.")
    parser.add_argument(
        "--show_plot",
        type=lambda x: bool(distutils.util.strtobool(x.strip())),
        default=True,
        help="Display the pixels in an additional matplotlib figure.",
    )
    parser.add_argument("--render_mode", type=str, choices=("depth", "rgb"), default="depth", help="")
    parser.add_argument(
        "--split_up_tiles",
        type=lambda x: bool(distutils.util.strtobool(x.strip())),
        default=True,
        help="Whether to split tiles into subplots when --show_plot is True.",
    )
    parser.add_argument("--custom_tile_arrangement", action="store_true", help="Apply custom tile arrangement.")

    args = parser.parse_known_args()[0]
    
    
    with wp.ScopedDevice(args.device):
        example = Example(num_tiles=args.num_tiles, custom_tile_arrangement=args.custom_tile_arrangement)
       
        
        while example.renderer.is_running():
            example.render()

        example.renderer.clear()

```

But I encountered the Error:

[image]

Env:

OS: ubuntu22.04
CUDA: 12.4    
Driver: 550.90.07

shi-eric, 2024-12-25:
Can you please paste the output of `wp.init()`? 

Modifying the example to set `wp.config.verify_cuda = True` can also give us more information to make sure that the problem lies in `cuda_graphics_register_gl_buffer()`

shi-eric, 2025-02-02:
Closing for lack of activity.

closed 2025-02-02 by shi-eric

shi-eric, 2025-10-29:
> Currently, I have the version `580.95.05` and I still have the same problem. Could be something related to the warning? The `CUDA/OpenGL interoperability is not available`

Yes, but I'm not quite sure why it's happening on your system. I have the same version of the driver and Warp nightly build installed, but `python warp/examples/core/example_render_opengl.py` runs fine.