NVIDIA/warp #407: [QUESTION] How to calculate the gradients of different parameters from the same kernel

Eigenoperator, 2024-12-26:
I am trying to calculate the gradients by using tape = wp.Tape() and tape.backward. Everything works just right when I only call the tape.backward once.  However, when I do several calculations in a single kernel and I try to calculate the gradients of several different parameters, the does NOT work as usual.
Here is an example.

import warp as wp
import numpy as np

@wp.kernel
def double(x: wp.array(dtype=float), y: wp.array(dtype=float), z: wp.array(dtype=float), s: wp.array(dtype=float)):
    tid = wp.tid()
    y[tid] = x[tid] 
    z[tid] = y[tid] 
    s[tid] = z[tid]

x = wp.array(np.arange(3), dtype=float, requires_grad=True)
y = wp.zeros_like(x)
z = wp.zeros_like(x)
s = wp.zeros_like(x)

tape = wp.Tape()
with tape:
    wp.launch(double, dim=3, inputs=[x, y, z, s])

tape.backward(grads={y: wp.ones_like(x)})
print(x.grad)
tape.zero()

tape.backward(grads={z: wp.ones_like(x)})
print(x.grad)
tape.zero()

tape.backward(grads={s: wp.ones_like(x)})
print(x.grad)

After running this code, it prints 
[1. 1. 1.]
[2. 2. 2.]
[4. 4. 4.].

But the gradients of y, z and s are supposed to be [1. 1. 1.]. So my main question is how to calculate the gradients of different parameters from the same kernel?

shi-eric, 2024-12-30:
Thanks for reporting this issue. I can try to explain the reason for the behavior you're seeing. The issue is that the `grads` argument will cause the tape to add the arrays to a set of gradients that are not reset to zero when `tape.zero()` is called. It seems that the `grads`  argument is a legacy feature according to the comment here:

https://github.com/NVIDIA/warp/blob/5b7a93415784350599b91ad28c9aba89e12b624b/warp/tape.py#L99-L109

We can see this in action by printing out `y.grad` after the first time you call `tape.zero()`. Instead of all 0's, `y.grad` is still full of 1's.

For the situation of calculating multiple gradients from the same tape, I think you can use the `fill_()` method of a gradient array to set it to 1, for example:

```python
y.grad.fill_(1.0)
tape.backward()
print(x.grad)
tape.zero()

z.grad.fill_(1.0)
tape.backward()
print(x.grad)
tape.zero()

s.grad.fill_(1.0)
tape.backward()
print(x.grad)
```

This results in the output:

```
[1. 1. 1.]
[1. 1. 1.]
[1. 1. 1.]
```

@daedalus5 do you remember the reason for 3845fbf?

daedalus5, 2025-01-02:
`const_gradients` is a legacy feature based on how the `grads` parameter was used in the past. I think it can be safely removed.

closed 2025-01-08 by daedalus5