NVIDIA/warp #408: [BUG] Warp CUDA error

robotNo201, 2024-12-27:
### Bug Description

isn't that funny?
[image]
[image]
[image]

### System Information

[image]
[image]
[image]

shi-eric, 2024-12-30:
I'm not sure what's going on and I can't reproduce an issue on my system, but have you tried installing `warp-lang` from PyPI? From the image you posted, it seems you installed [warp_lang-1.5.0+cu11-py3-none-manylinux2014_x86_64.whl](https://github.com/NVIDIA/warp/releases/download/v1.5.0/warp_lang-1.5.0+cu11-py3-none-manylinux2014_x86_64.whl), which works on my system with driver `560.35.03`. But I guess we might get more information if you run into the same issue with the `warp-lang` on PyPI.

FYI, Warp actually doesn't use the `nvcc` in your system path unless you are building Warp from source.

There was a recent commit from @nvlukasz on the `main` branch that fixed an issue with driver entry points, but it was meant to fix something else: 2df89edcb0bccbb248dc2a3f17bea24dee08dde9 Might be worth building Warp from the `main` branch on your local system and seeing what happens.

nvlukasz, 2025-01-02:
Interesting.  Looks like we fail to get the `cuGetProcAddress()` function from the driver. This function is needed to get all required driver entry points, including `cuDriverGetVersion()`. Since we fail to get that function, the detected driver version falls back to 0.0.

It's unclear why `cuGetProcAddress()` is not found, it should be present in the driver.

Based on nvidia-smi, the driver is 561.09, which I believe is a Windows driver. The kernel cache dir looks Linux-like (`/home/.../.cache/warp/1.5.0`). Is it running on WSL?

nvlukasz, 2025-01-02:
FWIW, I tried to reproduce this with WSL but didn't see a problem.

[image]

The nvidia-smi and driver versions match yours:
[image]

There might be an issue with your driver installation or WSL setup. I suggest to update your driver to latest (try custom installation option with clean install).  And like Eric suggested, maybe try the latest PyPI build of Warp:
```
pip install --upgrade --force-reinstall warp-lang
```

If that still fails, I the issue might be related to WSL somehow.

nvlukasz, 2025-01-02:
Just to get another data point, you could try running some Torch CUDA code in that environment.

test_torch.py:
```python
import torch
a = torch.arange(10, device="cuda:0")
print(a)
```

```console
$ pip install torch
$ python test_torch.py
```

See if that works correctly.

shi-eric, 2025-02-02:
Closing for lack of activity.

closed 2025-02-02 by shi-eric