NVIDIA/warp #409: [QUESTION] <title> Visualize the calculated result

sxn22012023, 2024-12-27:
Hi! I am re-running granular collision sdf example. But when I imported the calculated usd file into Blender, there is no collision between rock and particles? Can you tell me how I should visualize the calculated result?

**These are the results I imported into Blender.**
[image]

[image]

**This is the result from the official documentation.** 

[image]

shi-eric, 2024-12-30:
Thanks for reporting this issue. It seems that when 02d8d335ef1422040685d8cad09c223e30617542 adjusted the scale and orientation of the example assets, the granular collision SDF was not fully updated. This led to the rock collision geometry being a factor of 100x smaller than what's shown in the example gallery image. Try out this updated example:

```python
# Copyright (c) 2022 NVIDIA CORPORATION.  All rights reserved.
# NVIDIA CORPORATION and its licensors retain all intellectual property
# and proprietary rights in and to this software, related documentation
# and any modifications thereto.  Any use, reproduction, disclosure or
# distribution of this software and related documentation without an express
# license agreement from NVIDIA CORPORATION is strictly prohibited.

###########################################################################
# Example Sim Granular Collision SDF
#
# Shows how to set up a particle-based granular material model using the
# wp.sim.ModelBuilder(). This version shows how to create collision geometry
# objects from SDFs.
#
# Note: requires a CUDA-capable device
###########################################################################

import os

import numpy as np

import warp as wp
import warp.examples
import warp.sim
import warp.sim.render

class Example:
    def __init__(self, stage_path="example_granular_collision_sdf.usd"):
        fps = 60
        self.frame_dt = 1.0 / fps

        self.sim_substeps = 64
        self.sim_dt = self.frame_dt / self.sim_substeps
        self.sim_time = 0.0

        self.radius = 0.1

        builder = wp.sim.ModelBuilder()
        builder.default_particle_radius = self.radius

        builder.add_particle_grid(
            dim_x=16,
            dim_y=32,
            dim_z=16,
            cell_x=self.radius * 2.0,
            cell_y=self.radius * 2.0,
            cell_z=self.radius * 2.0,
            pos=wp.vec3(0.0, 20.0, 0.0),
            rot=wp.quat_identity(),
            vel=wp.vec3(2.0, 0.0, 0.0),
            mass=0.1,
            jitter=self.radius * 0.1,
        )
        with open(os.path.join(warp.examples.get_asset_directory(), "rocks.nvdb"), "rb") as rock_file:
            rock_vdb = wp.Volume.load_from_nvdb(rock_file.read())

        rock_sdf = wp.sim.SDF(rock_vdb)

        builder.add_shape_sdf(
            ke=1.0e4,
            kd=1000.0,
            kf=1000.0,
            mu=0.5,
            sdf=rock_sdf,
            body=-1,
            pos=wp.vec3(0.0, 0.0, 0.0),
            rot=wp.quat(0.0, 0.0, 0.0, 1.0),
[... 101 lines cut ...]
if __name__ == "__main__":
    import argparse

    parser = argparse.ArgumentParser(formatter_class=argparse.ArgumentDefaultsHelpFormatter)
    parser.add_argument("--device", type=str, default=None, help="Override the default Warp device.")
    parser.add_argument(
        "--stage_path",
        type=lambda x: None if x == "None" else str(x),
        default="example_granular_collision_sdf.usd",
        help="Path to the output USD file.",
    )
    parser.add_argument("--num_frames", type=int, default=400, help="Total number of frames.")

    args = parser.parse_known_args()[0]

    with wp.ScopedDevice(args.device):
        example = Example(stage_path=args.stage_path)

        for _ in range(args.num_frames):
            example.step()
            example.render()

        if example.renderer:
            example.renderer.save()
```

sxn22012023, 2025-01-03:
@shi-eric Thank you so much for your patient reply! The new code displays the normal size of the rock. However, when I imported the result into Blender and clicked the play button, nothing happened. The entire result is static, and the granules can’t fall onto the rock. Is there something I did wrong?

[image]

shi-eric, 2025-01-03:
I don't think you did anything wrong; it could be an issue with Blender's USD import. Here's a screenshot from opening the file in [usdview](uhttps://docs.omniverse.nvidia.com/usd/latest/usdview/quickstart.htmlrl):

[image]

@christophercrouzet Do you have any insight about the Blender issue?

christophercrouzet, 2025-01-05:
Hi @sxn22012023, I managed to visualize the expected result in USD View, Omniverse, and Houdini, but I can repro your issue in Blender 4.3.2.

From what I can tell, everything is set correctly during the import, and the frame range is correct, so this _should_ work. My first guess would also be some possible shortcomings with Blender's USD importer.

christophercrouzet, 2025-01-07:
Hi @sxn22012023, the import issue has been addressed in Blender 4.4 alpha, which you can try [here](https://builder.blender.org/download/daily/).

sxn22012023, 2025-01-09:
@christophercrouzet @shi-eric Thank you so much to both of you!!! I have tested the Blender 4.4 alpha!! It really works! Thanks again!!

closed 2025-01-18 by shi-eric