NVIDIA/warp #414: [BUG] Cannot find Prim Path on USD File Generated from USD Renderer

rrzhang139, 2025-01-03:
### Bug Description

Hello,

I am trying to access the vertices and tets of a USD mesh that was generated from warp's USD renderer. But I am having trouble finding the correct prim path. Here is a simple script I wrote that lists all the paths and prints all the mesh's vertices. 

```
from pxr import Usd, UsdGeom

def peek_inside_usd(usd_path):
    """Print all available prim paths inside a USD file"""
    try:
        stage = Usd.Stage.Open(usd_path)
        print(f"\nPaths available in {usd_path}:")
        print("------------------------")
        for prim in stage.Traverse():
            # Print path and type of prim
            print(f"{prim.GetPath()} ({prim.GetTypeName()})")
            
            # If it's a mesh, print additional info
            if prim.GetTypeName() == "Mesh":
                mesh = UsdGeom.Mesh(prim)
                points = mesh.GetPointsAttr().Get()
                if points is not None:
                    print(f"  - Vertices: {len(points)}")
                    print(f"  - This is a valid mesh path!")
            elif prim.GetTypeName() == "TetMesh":
                mesh = UsdGeom.TetMesh(prim)
                points = mesh.GetPointsAttr().Get()
                if points is not None:
                    print(f"  - Vertices: {len(points)}")
                    print(f" - Tet indices: {len(mesh.GetTetVertexIndicesAttr().Get())}")
                    print(f"  - This is a valid tet mesh path!")
            elif prim.GetTypeName() == "PointInstancer":
                mesh = UsdGeom.PointInstancer(prim)
                points = mesh.GetPositionsAttr().Get()
                print(points)
                if points is not None:
                    print(f"  - Vertices: {len(points)}")
                    # print(f" - Tet indices: {len(mesh.GetTetVertexIndicesAttr().Get())}")
                    print(f"  - This is a valid tet mesh path!")
        print("------------------------\n")
    except Exception as e:
        print(f"Error opening {usd_path}: {e}")

usd_file = "assets/test_target.usd"
peek_inside_usd(usd_file)
```

Simply insert any usd file in `usd_file`. 

Here was my output: 
```
/root (Xform)
/root/particles (PointInstancer)
None
/root/particles/sphere (Sphere)
/root/surface (Mesh)
/dome_light (DomeLight)
/distant_light (DistantLight)
```

christophercrouzet, 2025-01-08:
Hi @rrzhang139, I'm not clear what your question is but the answer likely depends on the contents of the USD scene.

For example, here's how to retrieve the the mesh geometry from the bunny asset shipped with Warp's examples:

```py
from pxr import Usd, UsdGeom

stage = Usd.Stage.Open("./warp/examples/assets/bunny.usd")
prim = stage.GetPrimAtPath("/root/bunny")
mesh = UsdGeom.Mesh(prim)
print(mesh)
# UsdGeom.Mesh(Usd.Prim(</root/bunny>))
```

Does this help?

shi-eric, 2025-02-02:
Closing for lack of activity.

closed 2025-02-02 by shi-eric