NVIDIA/warp #417: [REQUEST] Update example_walker.py to use tile API

shi-eric, 2025-01-07:
### Description

`wp.matmul()` will eventually be removed, so we want to make sure this example uses the tile programming model.

shi-eric, 2025-01-07:
BTW, one complication is that the tile API doesn't work with CUDA 11 builds due to the use of nvJitLink...

shi-eric, 2025-01-08:
There's also a usage of `wp.matmul()` in the linear solvers module.

gdaviet, 2025-01-08:
Yes, `wp.matmul` is currently used as the default dense matrix-vector multiplication algorithm for the iterative linear solvers. A tile-based alternative should be possible, and I don't believe the linear solvers are part of the officially supported API, so limiting support to cuda 12 might be an option.

closed 2025-01-17 by shi-eric