NVIDIA/warp #420: [BUG] Investigate Inconsistencies With Array Shape and Strides

christophercrouzet, 2025-01-09:
### Bug Description

Test case 1:

```py
import torch
import warp as wp

# Setting the device to CUDA yields different results.
device = "cpu"

with wp.ScopedDevice(device):
    torch_device = wp.device_to_torch(device)

    torch_arr = torch.tensor(
        (
            ( 1,  2,  3,  4),
            ( 5,  6,  7,  8),
            ( 9, 10, 11, 12),
            (13, 14, 15, 16),
            (17, 18, 19, 20),
            (21, 22, 23, 24),
        ),
        dtype=torch.int32,
        device=torch_device,
    )
    assert torch_arr.size() == (6, 4)
    assert torch_arr.stride() == (4, 1)

    warp_arr = wp.array(torch_arr)
    assert warp_arr.shape == (6, 4)
    assert warp_arr.strides == (16, 4)

    torch_arr = torch.as_strided(
        torch_arr,
        size=torch_arr.size(),
        stride=(torch_arr.stride(1), torch_arr.stride(0)),
    )
    assert torch_arr.size() == (6, 4)
    assert torch_arr.stride() == (1, 4)

    arr_i32 = wp.array(torch_arr, dtype=wp.int32)
    arr_v2i = wp.array(torch_arr, dtype=wp.vec2i)
    print(f"arr_i32 strides: {arr_i32.strides}, expected: (4, 16), correct: {arr_i32.strides == (4, 16)}")
    print(f"arr_v2i strides: {arr_v2i.strides}, expected: (8, 16), correct: {arr_v2i.strides == (8, 16)}")

# cpu:
# arr_i32 strides: (4, 16), expected: (4, 16), correct: True
# arr_v2i strides: (16, 8), expected: (8, 16), correct: False

# cuda:
# arr_i32 strides: (4, 16), expected: (4, 16), correct: True
# arr_v2i strides: (16, 8), expected: (8, 16), correct: False
```

Test case 2:

```py
import warp as wp

with wp.ScopedDevice("cuda"):
    array = wp.array(((1, 2), (3, 4), (5, 6)), dtype=wp.vec2s)
    view = wp.array(array, dtype=wp.vec3s)
    assert view.dtype is wp.vec3s
    print(f"shape: {view.shape}, expected: (2,), correct: {view.shape == (2,)}")
    print(f"strides: {view.strides}, expected: (6,), correct: {view.strides == (6,)}")

# RuntimeError: The inner dimensions of the input data are not compatible with the requested vector type vec3s: expected an inner dimension that is a multiple of 3
```

Test case 3:

```py
import warp as wp

with wp.ScopedDevice("cuda"):
    array = wp.array(((1, 2, 3), (4, 5, 6), (7, 8, 9)), dtype=wp.mat33d)
    view = wp.array(array, dtype=wp.vec3d)
    assert view.dtype is wp.vec3d
    print(f"shape: {view.shape}, expected: (3,), correct: {view.shape == (3,)}")
    print(f"strides: {view.strides}, expected: (24,), correct: {view.strides == (24,)}")

# shape: (1, 3), expected: (3,), correct: False
# strides: (72, 24), expected: (24,), correct: False
```

0z5a, 2026-09-19:
@christophercrouzet Hi, I'd be interested in taking a first pass at this issue.

christophercrouzet, 2026-09-28:
Thanks @0z5a. This kind of issue can be a bit tricky, so I'd rather we iterate on these internally.