NVIDIA/warp #432: [BUG] add_builder does not take into account body_count

jc211, 2025-01-14:
### Bug Description

Calling ```builder.add_builder(articulation_builder)``` does not offset the ```builder.joint_parent``` and ```builder.joint_child``` by ```self.body_count```. As a result, the bodies of the links merge with bodies already existing in builder

jc211, 2025-02-14:
https://github.com/NVIDIA/warp/blob/cde8deaa5d1b13b5421bfec71902f73fe4a06ac4/warp/sim/model.py#L1424-L1425
These should be 
```python
            self.joint_parent.extend([p + self.body_count if p != -1 else -1 for p in builder.joint_parent])
            self.joint_child.extend([c + self.body_count for c in builder.joint_child])
```

shi-eric, 2025-02-20:
Thanks for reporting, this was addressed by @eric-heiden in 66495e73c637e1beeba2cfa43287bc7356fb8018

closed 2025-02-20 by shi-eric