NVIDIA/warp #444: [QUESTION] <title>

Liudeke, 2025-01-21:
There are no examples of XPBDIntegrator software simulations

shi-eric, 2025-01-21:
Can you be more specific of what you're looking for? A quick search in the repository shows the following examples that use `wp.sim.XPDIntegrator`:

- https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_cloth.py
- https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_particle_chain.py
- https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_quadruped.py
- https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_rigid_force.py

Liudeke, 2025-01-21:
> Can you be more specific of what you're looking for? A quick search in the repository shows the following examples that use `wp.sim.XPDIntegrator`:
>
> * [...]

But there are no examples of softbody simulations

MontaEllis, 2025-01-22:
Same issue. Can you provide some samples of garment simualtion?

sdasika06, 2025-02-04:
I would also like to see an example of a soft_grid simulation with the XPBD Integrator constraints.