NVIDIA/warp #455: [BUG] Reduce xpbd torsional stiffness default

jc211, 2025-01-24:
### Bug Description

```model.rigid_contact_torsional_friction = 0.5``` is quite high as a default and does not allow natural movements to occur in XPBD.

shi-eric, 2025-01-28:
Changing model defaults is relatively risky since it can affect many users, and we will likely have a better integrator for rigid body dynamics coming later this year. You can still manage for now by overriding the default value, right?

jc211, 2025-01-28:
Yeah that's easy enough - its just if anyone tries to use this rigid body simulator they will quickly find that objects that should rotate when pushed do not and its this particular setting that is causing this behaviour.

closed 2025-11-06 by shi-eric