NVIDIA/warp #460: [BUG] Running warp.examples.core.example_torch, Matplotlib set_data expected sequence

ped02, 2025-01-26:
### Bug Description

**Bug Description**
After installing warp, I tried to check the setup by running examples. I am having a problem running a few of them. Specifically for warp.examples.core.example_torch. 

```
File "/home/user/python_envs/warp_test/lib/python3.10/site-packages/warp/examples/core/example_torch.py", line 186, in step_and_render
    self.render()
  File "/home/user/python_envs/warp_test/lib/python3.10/site-packages/warp/examples/core/example_torch.py", line 179, in render
    self.mean_marker.set_data(self.mean_pos[0], self.mean_pos[1])
  File "/home/user/python_envs/warp_test/lib/python3.10/site-packages/matplotlib/lines.py", line 666, in set_data
    self.set_xdata(x)
  File "/home/user/python_envs/warp_test/lib/python3.10/site-packages/matplotlib/lines.py", line 1290, in set_xdata
    raise RuntimeError('x must be a sequence')
```
I'm not sure this is due to Matplotlib dependency version or if its an actual bug.
Matplotlib version: 3.10.0

Location:
https://github.com/NVIDIA/warp/blob/2961bd096be69e261e98d8609bd1066ea839fe41/warp/examples/core/example_torch.py#L179C9-L179C70

Work-around
Added [] in the argument
```
self.mean_marker.set_data([self.mean_pos[0]], [self.mean_pos[1]])
```

**Installation Method**
Virtual Environment Python 3.10
`pip install warp-lang[extras]`

**System Information**
Warp Version: 1.5.1
CUDA Version: 12.5
OS Ubuntu 22.04
Python Version: 3.10.12

shi-eric, 2025-01-27:
Thanks for the bug report! This will be fixed.

shi-eric, 2025-01-28:
This bug has been fixed in 9abb04e.

closed 2025-01-28 by shi-eric