NVIDIA/warp #467: [QUESTION] cuda graph and sim.collide ?

ucacaxm, 2025-01-29:
Hi,

When I activate the cuda graph, the call to sim.collide generates an error.
My code for the creation of the graph:
```
       self.use_cuda_graph = wp.get_device().is_cuda
       if self.use_cuda_graph:
            with wp.ScopedCapture() as capture:
                self.loss.zero_()
                self.tape = wp.Tape()
                self.tape.zero()
                with self.tape:
                    self.forward()
                self.tape.backward(self.loss)
            self.graph = capture.graph
```
and the `forward` function:
```
        self.states[0] = self.model.state(requires_grad=True)
        for i in range(self.num_frames):
            wp.sim.collide(self.model, self.states[i])
            self.states[i].clear_forces()
            wp.launch(apply_action, self.action_dim, inputs=[self.actions, i * self.action_dim], outputs=[self.control.joint_act])
            self.states[i + 1] = self.integrator.simulate(self.model, self.states[i], self.states[i+1], self.sim_dt, self.control)
        wp.launch( compute_loss, dim=1, inputs=[self.states[-1].body_q], outputs=[self.loss])
```
I use the autodiff to update the array `self.actions`.
The error during the creation of the graph:
```
Warp CUDA error 1: invalid argument (in function memset_device, C:\g\131577751\warp\native\warp.cu:826)
Warp CUDA error 1: invalid argument (in function memcpy_d2d, C:\g\131577751\warp\native\warp.cu:701)
```

If I comment the call to `collide` I don't have the error anymore, but the object cross the floor (which is normal).

Is there something I missed ?

Thanks,

PS: my setup
```
Warp 1.5.1 initialized:
   CUDA Toolkit 12.6, Driver 12.4
   Devices:
     "cpu"      : "Intel64 Family 6 Model 170 Stepping 4, GenuineIntel"
     "cuda:0"   : "NVIDIA RTX 1000 Ada Generation Laptop GPU" (6 GiB, sm_89, mempool enabled)
```

shi-eric, 2025-01-29:
It's hard for me to suggest what might be wrong from the information provided; I see that we have some examples that use `wp.sim.collide()` inside a graph, so it might be worthwhile to compare what you're doing differently from those examples. For example, maybe don't do `self.states[0] = self.model.state(requires_grad=True)`  in  `forward()`. You can also simplify `self.states[i + 1] = self.integrator.simulate(self.model, self.states[i], self.states[i+1], self.sim_dt, self.control)` to `self.integrator.simulate(self.model, self.states[i], self.states[i+1], self.sim_dt, self.control)`.

ucacaxm, 2025-01-30:
Thanks for you help.

I checked different examples before posting. 
The strange thing I have is even with `collide` commented, I need to call once `forwad()` before the construction of the graph, otherwise I get the same error. It is like the CUDA arrays need to be created before the graph.
Maybe I have something with the model builder ...

My full code
```
import math
import os
import numpy as np
import warp as wp
import warp.examples
import warp.sim
import warp.sim.render
from warp.optim import Adam
import pyglet

@wp.kernel
def compute_loss(
    bq: wp.array(dtype=wp.transform), 
    loss: wp.array(dtype=wp.float32)
):
    v = wp.vec3( bq[0][0] - bq[2][0],  0.0,  bq[0][2] - bq[2][2])
    loss[0] = 3.0-bq[0][1] + bq[2][1] + wp.dot(v, v)

@wp.kernel
def apply_action(action_src: wp.array(dtype=wp.float32), start_index: int, action_dst: wp.array(dtype=wp.float32)):
    i = wp.tid()
    action_dst[i] = action_src[start_index + i]

class MocapTracking:
    def __init__(self, num_envs=1):
        builder = wp.sim.ModelBuilder()

        wp.sim.parse_mjcf(
        #custom_parse_mjcf(
            # "data/mujoco/sikax.xml",
            "data/mujoco/oneLeg.xml",
            #"data/mujoco/biped.xml",
            # "data/mujoco/ant.xml",
            # "data/mujoco/smpl.xml",
            #"data/mujoco/simple.xml",
            builder,
            xform=wp.transformf(wp.vec3f(0, 1.5, 0), wp.quat_from_axis_angle(wp.vec3f(0, 0, 1), -math.pi/2)),
            armature=2.0,
            stiffness=500.5,
            damping=15.0,
            contact_ke=1.0e4,
            contact_kd=1.0e2,
            contact_kf=1.0e2,
            contact_mu=0.75,
            limit_ke=1.0e4,
            limit_kd=1.0e1,
            up_axis="Z",
        )

        self.with_control = True #False

        self.sim_time = 0.0
        self.fps = 100
        self.sim_dt = 1.0 / self.fps

        # actions
        builder.joint_q[-3:] = [  1.6, 0.3, 0.4]    # random value to test the model. The array has 7 values (R+T global) + the 3 values (joints) 
        builder.joint_axis_mode = [wp.sim.JOINT_MODE_TARGET_POSITION] * len(builder.joint_axis_mode)    # define mode for joint_act

        self.opti_count = 0
        self.opti_duration_in_sec = 1.0
        self.num_frames = int(self.opti_duration_in_sec * self.fps)
[... 83 lines cut ...]
        self.opti_count += 1

    def optiNStep(self, n):
        for i in range(n):
            self.stepOpti()
            if self.opti_count % 10 == 0:
                loss = self.loss.numpy()
                print(f"Iteration {self.opti_count}: {loss}")

    def renderOneSequence(self):
        for i in range(self.num_frames*10):            # render 10 times the same frame (to slow down the animation)
            self.renderer.begin_frame(self.sim_time)
            self.renderer.render(self.states[ int(i/10) ])
            self.renderer.end_frame()

    def run(self, device=None):
        with wp.ScopedDevice(device):
            self.forward()
            while self.renderer.is_running():
                self.renderOneSequence()

if __name__ == "__main__":
    viewer = MocapTracking()
    viewer.run()
```

ucacaxm, 2025-01-30:
My fault, in `run`I also have to use the graph:
(and I had a mess with the devices)

```
   def run(self):
        if self.use_cuda_graph:
            wp.capture_launch(self.graph)
        else: 
            self.forward()
        while self.renderer.is_running():
            self.renderOneSequence()
```

Thanks for your help

closed 2025-01-30 by ucacaxm

shi-eric, 2025-06-15:
> Can you explain why this make a difference, please? I am having the same trouble with collide being inside a graph.

Can you file a new topic with an example? I also want to make sure you're using a CUDA driver from 12.3 and onwards to avoid any issues with compiling and loading modules during a graph capture.

DINHQuangDung1999, 2025-06-15:
@shi-eric It seems that I basically make the same mistake with OP.  The problem was me trying to define another function called `forward1()` that is exactly the same as `forward()`, except that it was not captured into CUDA graph. I believe since the memory is already allocated for the graph, calling `forward1()` causes an error.  When I launch the graph instead of calling `forward1()`, it runs perfectly.