NVIDIA/warp #471: [BUG] Exploding Gradients for force-control of robot arm

JonathanKuelz, 2025-01-29:
### Bug Description

Hey warp team, thanks for providing this library!

**Scenario**: I am trying to force-control a robot, loaded from URDF file, to reach a goal position after `n` timesteps.

**Approach**: Initialize with small joint forces and run the simulation, then backpropagate the loss and use the gradient information to update the joint forces.

**Bug**: The gradients quickly explode. After few timesteps already, they go towards infinity before finally becoming `nan`.

The [Frankie](https://github.com/Daniella1/urdf_files_dataset/blob/main/urdf_files/robotics-toolbox/xacro_generated/franka_description/robots/frankie.urdf) urdf I use is public, but this code should run with any URDF (maybe it won't converge, but that's not the issue at hand). Here is some more detailed information:

<details>
<summary>1) The code I use. It should run with any URDF. </summary>

```python
from pathlib import Path

import numpy as np
import warp as wp
import warp.optim
import warp.sim
import warp.sim.render

np.set_printoptions(
    precision=3,
    linewidth=120
)

@wp.kernel
def apply_control(control: wp.array(dtype=wp.float32),
                  at_iter: int,
                  action_dim: int,
                  joint_torques: wp.array(dtype=wp.float32)):
    start_index = at_iter * action_dim
    i = int(0)
    for idx in range(start_index, start_index + action_dim):
        joint_torques[i] =  control[idx]
        i = i + 1

@wp.kernel
def loss_kernel(reached: wp.array(dtype=wp.transform), body_id: int, goal: wp.vec3, loss: wp.array(dtype=float)):
    diff = wp.transform_get_translation(reached[body_id]) - goal
    l = wp.dot(diff, diff)
    loss[0] = l

class RobotSim:

    def __init__(self,
                 stage_path: str | Path = "frankie_force_control.usd",
                 render: bool = True,
                 substeps: int = 5,
                 num_frames: int = 300,
                 fps: int = 30,
                 reach_goal: wp.vec3 = wp.vec3(1., 0.5, -0.2),
                 verbose: bool = False,
                 ):
        self.verbose: bool = verbose
        builder = wp.sim.ModelBuilder()

        warp.sim.parse_urdf(
            "<path_to_frankie_urdf>",
            builder,
            floating=False,
            xform=wp.transform([0.0, 0.0, 0.0], wp.quat_from_axis_angle(wp.vec3(1.0, 0.0, 0.0), -np.pi * 0.5)),
            density=1000,
            armature=0.1,
            stiffness=100,
            damping=40,
            enable_self_collisions=False  # TODO: Bug in warp, see https://github.com/NVIDIA/warp/issues/441
[... 123 lines cut ...]
These are the action gradients if I run the simulation for **2 frames only**.
```
[ 0.000e+00  5.434e+32 -5.434e+32 -8.481e+29 -3.798e+31  8.892e+30 -3.528e+31  1.772e+30  9.507e+31 -3.113e+31
 -2.099e+29  0.000e+00 -3.709e+28  3.709e+28  4.751e+25  2.202e+27 -5.176e+26  2.060e+27 -9.130e+25 -5.567e+27
  1.570e+27  1.293e+25  0.000e+00  2.535e+24 -2.535e+24 -1.401e+21 -1.265e+23  2.938e+22 -1.166e+23  4.656e+21
  3.176e+23 -7.364e+22 -7.893e+20  0.000e+00 -1.734e+20  1.734e+20  6.317e+16  6.956e+18 -1.597e+18  6.470e+18
 -2.794e+17 -1.756e+19  3.152e+18  4.781e+16  0.000e+00  1.187e+16 -1.187e+16 -6.914e+12 -3.669e+14  8.718e+13
 -3.336e+14  1.099e+13  9.158e+14 -1.062e+14 -2.846e+12  0.000e+00 -8.125e+11  8.125e+11  1.480e+08  1.744e+10
 -3.877e+09  1.605e+10 -7.554e+08 -4.445e+10  1.364e+09  1.684e+08  0.000e+00  5.565e+07 -5.565e+07 -2.477e+04
 -7.311e+05  1.820e+05 -6.406e+05  3.690e+04  1.805e+06  1.911e+05 -1.751e+04  0.000e+00 -3.811e+03  3.811e+03
  3.219e-01  1.625e+01 -3.313e+00  1.197e+01  1.487e-01 -4.038e+01 -2.192e+01  3.332e+01  0.000e+00  2.024e-01
 -2.024e-01 -7.782e-03 -2.767e-02 -4.359e-03 -1.558e-02 -1.598e-04 -3.334e-02 -2.900e-04  1.355e+00  0.000e+00
  2.288e-03 -2.288e-03  0.000e+00  0.000e+00  0.000e+00  0.000e+00  0.000e+00  0.000e+00  0.000e+00  0.000e+00]
```

...as you can see, it's already filled with some huge numbers. From 3 frames on, its mostly nans.

</details>

Have I forgotten something? What should I do to obtain stable gradients for this supposedly simple scenario? I would appreciate any ideas :)

### System Information

- NVIDIA GeForce RTX 3080 Ti 
- warp version 1.5.0

MontaEllis, 2025-02-03:
same issue, have you solved it?

JonathanKuelz, 2025-02-03:
No, the problem still persists. I got slightly better results using the FeatherStoneIntegrator, but still not satisfactory -- the gradients remain higher than I would think them to be and eventually explode after a couple of steps.

Tried a couple of things, but I did not find a way to get meaningful gradients for a force-controlled URDF import using the SemiImplicitIntegrator

MontaEllis, 2025-02-03:
Me too. I also noticed that using default parameter values improves the gradient, but the results are unconvincing.