NVIDIA/warp #485: [BUG] Warp truncates double-precision constants

shi-eric, 2025-02-02:
### Bug Description

Double-precision constants get truncated to single-precision unless done in a specific manner.

## Example

```python
import warp as wp

@wp.kernel
def test_precision():
    pi = wp.float64(wp.PI)
    wp.printf("pi (kernel) = %.16g\n", pi)

print(f"pi (python) = {wp.PI}")

wp.launch(test_precision, dim=(1,), inputs=[])
```

This produces the output:

```
pi (python) = 3.141592653589793
pi (kernel) = 3.141592741012573
```

We can see what's going on in the generated CUDA C++ code:

```cpp
extern "C" __global__ void test_precision_d1c1f7f9_cuda_kernel_forward(
    wp::launch_bounds_t dim)
{
    for (size_t _idx = static_cast<size_t>(blockDim.x) * static_cast<size_t>(blockIdx.x) + static_cast<size_t>(threadIdx.x);
         _idx < dim.size;
         _idx += static_cast<size_t>(blockDim.x) * static_cast<size_t>(gridDim.x))
    {
        // reset shared memory allocator
        wp::tile_alloc_shared(0, true);

        //---------
        // primal vars
        const wp::float32 var_0 = 3.141592653589793;
        const wp::float32 var_1 = 3.141592653589793;
        wp::float64 var_2;
        const wp::str var_3 = "pi (kernel) = %.16g\n";
        //---------
        // forward
        // def test_precision():                                                                  <L 7>
        // pi = wp.float64(wp.PI)                                                                 <L 8>
        var_2 = wp::float64(var_1);
        // wp.printf("pi (kernel) = %.16g\n", pi)                                                 <L 9>
        printf(var_3, var_2);
    }
}
```

The right double-precision value is being captured in the kernel, but it's assigned to a `wp::float32` first, causing the truncation.

## Current Workaround

A workaround is to define the variable as a `wp::float64` outside the kernel:

```python
import warp as wp

pi_fp64 = wp.float64(wp.PI)

@wp.kernel
def test_precision():
    pi = wp.float64(pi_fp64)
    wp.printf("pi (kernel) = %.16g\n", pi)

print(f"pi (python) = {pi_fp64.value}")

wp.launch(test_precision, dim=(1,), inputs=[])
```

This produces the expected output:

```
pi (python) = 3.141592653589793
pi (kernel) = 3.141592653589793
```

The generated code in this case is:

```cpp
extern "C" __global__ void test_precision_ae841f27_cuda_kernel_forward(
    wp::launch_bounds_t dim)
{
    for (size_t _idx = static_cast<size_t>(blockDim.x) * static_cast<size_t>(blockIdx.x) + static_cast<size_t>(threadIdx.x);
         _idx < dim.size;
         _idx += static_cast<size_t>(blockDim.x) * static_cast<size_t>(gridDim.x))
    {
        // reset shared memory allocator
        wp::tile_alloc_shared(0, true);

        //---------
        // primal vars
        const wp::float64 var_0 = 3.141592653589793;
        wp::float64 var_1;
        const wp::str var_2 = "pi (kernel) = %.16g\n";
        //---------
        // forward
        // def test_precision():                                                                  <L 7>
        // pi = wp.float64(pi_fp64)                                                               <L 8>
        var_1 = wp::float64(var_0);
        // wp.printf("pi (kernel) = %.16g\n", pi)                                                 <L 9>
        printf(var_2, var_1);
    }
}
```

c0d1f1ed, 2026-01-29:
This turns out to be a more complicated than expected. Just writing stuff down to sort my thoughts:

- In Python all floating-point numbers (`float` type) are double-precision. Even though we have `wp.float32` and `wp.float64` to discern between single- and double-precision at the Warp level, there is no single-precision _literal_ representation.
- We have lots of existing kernel code like `x * 2.0` where `x` is single-precision and `2.0` is interpreted as single-precision. In contrast, CUDA would implicitly promote `x` to double-precision and perform a double-precision multiplication. So the standard recommends using the `f` suffix where applicable.
- Warp does not support any implicit conversion at the moment.

We have a few design options:

- Keep things as-is. The behavior makes sense when accepting that `float` means single-precision in kernels and double-precision in Python. Only explicit use of `wp.float32` and `wp.float64` ensures a fixed format. While consistent and tractable, it's too confusing.
- Interpret literals as double precision and implement implicit conversion, requiring a cast to `wp.float32` to stick to single precision arithmetic. While this would be closest to CUDA behavior, it's not a realistic option due to causing terrible performance in many cases.
- Treat values representable in single-precision as single-precision, otherwise double-precision. This would mean `2.0` is single-precision. `wp.PI` would be double-precision, and we could have a single-precision variant. The issue with this is that it's hard to tell for a given expression containing literals or constants whether it produces a single- or double-precision result.
- Treat float literals and constants as single-precision except when casting them with `wp.float64()`. Basically use the cast as the inverse of the f-suffix. This is tricky because in `wp.float64(2.0 / 3.0)` should the division be performed in single- or double-precision? The expectation is probably double-precision but this requires either performing constant folding as part of codegen or ensuring each literal gets stored in a `double` at the CUDA level.
- Treat float literals and constants as double-precision but when used in arithmetic involving single-precision variables automatically cast them to single-precision.

The last option is what NumPy does: [Promotion rules for Python scalars](https://numpy.org/neps/nep-0050-scalar-promotion.html). Essentially `float` is considered a "weak" type and it only becomes strongly typed to match the precision of the arithmetic operations. It makes a lot of sense from a developer's perspective, with minimal surprising behavior.

It might be feasible to implement this using a C++ `weak_float` class which can cast to a float or double based on context...

shi-eric, 2026-03-09:
The scope was reduced and the precision loss issue was addressed in 0d3e1ea177680033d21d59b3151e3b079f83d47a, planned for the 1.12.1 release.