NVIDIA/warp #486: [BUG] Import USD sets body inertia to np array sometimes, fails to build model

JonathanKuelz, 2025-02-03:
### Bug Description

Importing USD's with `wp.sim.import_usd` fails if some bodies have non-zero inertia and some bodies have zero inertia.

I believe the issue is that [here](https://github.com/NVIDIA/warp/blob/e76006537835c8751c0abb307e0d475c479c0197/warp/sim/import_usd.py#L768), the inertia for some bodies (those with non-zero diagonal inertia) is computed as numpy array, whereas everywhere else, the inertia is computed as `wp.mat33`. This leads to a mix of datatypes in `builder.body_inertia` and eventually a RunTimeError [here](https://github.com/NVIDIA/warp/blob/e76006537835c8751c0abb307e0d475c479c0197/warp/sim/model.py#L4650).

Explicitly casting the inertia to `wp.mat33` in the import_usd function should be a simple fix as far as I can tell.

eric-heiden, 2025-02-03:
Thanks for reporting this issue @JonathanKuelz! Are you able to share a USD file with us we could utilize for testing these zero/nonzero inertia definition issues?

JonathanKuelz, 2025-02-03:
Unfortunately not really.

I was using the USD file provided [here](https://git.openlogisticsfoundation.org/silicon-economy/simulation-model/evobotsimmodel), but I had to apply another couple minor patches to warp before this issue even surfaced. I'm happy to summarize them in short below, but I don't know if these "problems" were by design, so I just reported the one about the inertia matrices, which doesn't seem to be intended. Even with my patches, some orientation information in the original USD seems to be lost still - the USD is loaded perfectly fine using isaac sim however.

Other patches I applied to warp to load the evoBot USD:
1. `joint.setdefault("parent", None)` before [this line](https://github.com/NVIDIA/warp/blob/e76006537835c8751c0abb307e0d475c479c0197/warp/sim/import_usd.py#L452) (parsing USD failed if the joint prim didn't explicitly define the world frame as parent frame)
2. Implemented a custom sort function for the children refs [here](https://github.com/NVIDIA/warp/blob/e76006537835c8751c0abb307e0d475c479c0197/warp/sim/import_usd.py#L572). Right now, warp cannot parse a USD if two bodies are children of the same parent prim but connected with a joint (Assume body `a` & `b` are both children of the same prim, and there is a joint that defines `b` as parent and `a` as child. `path_body_map` in `parse_prim` will not contain `b` at the time of parsing `a` and [this line](https://github.com/NVIDIA/warp/blob/e76006537835c8751c0abb307e0d475c479c0197/warp/sim/import_usd.py#L457) will throw a KeyError.
3. Lastly, the USD contains a prim with an almost-zero mass (at `</evoBOT/base_footprint>`, mass `1e-14`, I didn't investigate where that comes from) with a body with zero mass, so [this](https://github.com/NVIDIA/warp/blob/e76006537835c8751c0abb307e0d475c479c0197/warp/sim/import_usd.py#L752) failed. I guess the issue here is with the USD, I hotfixed it by replacing `builder.body_mass[body_id] = 0` with `builder.body_mass[body_id] < 1e-12` in the if-clause above.

If you follow these 3 steps, the inertia issue will finally arise.