NVIDIA/warp #492: [BUG] Builder: shape_body and shape_transform not same shape

jc211, 2025-02-03:
### Bug Description

```shape_body``` and ```shape_transform``` are not guaranteed to be the same shape when using ```add_builder``` https://github.com/NVIDIA/warp/blob/b7e4624c074632f29ec5bd953948ef689b55b700/warp/sim/model.py#L1400C24-L1400C25

As a result the ```CreateSimRenderer.populate``` will fail here: https://github.com/NVIDIA/warp/blob/b7e4624c074632f29ec5bd953948ef689b55b700/warp/sim/render.py#L160

jc211, 2025-02-04:
In general, it seems ```add_builder``` is not that usable, it seems to not work very well when the intention is to create a builder per environment. Its better off manually settings collision groups in the shapes.

JonathanKuelz, 2025-02-04:
Just out of curiosity, would you be willing to share your code? I am currently trying to set up parallel simulation using `parse_urdf`, but I'm having issues with severe instabilities for everything but the FeatherStoneIntegrator. I would be curious to see how you set up your environment.

eric-heiden, 2025-04-01:
The bug where `ModelBuilder.shape_transform` could end up having the wrong size has been fixed in https://github.com/NVIDIA/warp/commit/fb9fa494b3fa317bf8ea6f954ff3b748886076ab. The `shape_body` should also have the correct size after calling `add_builder`.

closed 2025-04-30 by shi-eric