NVIDIA/warp #496: [DOCS] Parsing and simulating URDFs/rigid body trees

JonathanKuelz, 2025-02-05:
## Category
- [ ] Report an error in the documentation.
- [x] Request for something to be documented.
- [ ] Suggestion to improve the documentation.
- [ ] Other (please explain)

## Description
Dear warp team,

it would be great to have more documentation and/or an example on how to use `warp.sim.parse_urdf`. Specifically, I couldn't find the following information:
- Calling `warp.sim.eval_fk` is strictly necessary before starting the simulation to update the body positions in space *if* using `SemiImplicitIntegrator` or `XPBDIntegrator`.
- On the other hand, it seems like `warp.sim.eval_ik` is necessary if `FeatherStoneIntegrator` is used and initial position/offset information for "floating" robots/bodies shouldn't get lost.
- Does the `SemiImplicitIntegrator` even work for kinematic trees of rigid bodies? Following the "standard setup", every URDF I have parsed so far behaved unstable unless I used the featherstone integrator. Below is a minimal example that fails, including some of the first frames of the simulation. The robot "explodes" (e.g., base and first link disconnect) and bodies do not stay connected. This happens across various URDFs. I tried tuning stiffness and damping parameter of the attachment constraints, but did not find a set that made it work.

Long story short: I have spent quite some time trying to make warp work for simulation of various robots, but so far have had limited success. It would be great to have some more documentation on how to make this work, since the functionalities provided by warp seem to have great potential for this area of application.

```python
from pathlib import Path

import numpy as np
import warp as wp
import warp.optim
import warp.sim
import warp.sim.render

URDF_PATH = "<ENTER_URDF_PATH_HERE>"

class RobotSim:

    def __init__(self,
                 stage_path: str | Path = "franka_urdf.usda",
                 render: bool = True,
                 substeps: int = 2,
                 num_frames: int = 240,
                 fps: int = 60,
                 verbose: bool = False,
                 integrator: str = 'featherstone'
                 ):
        self.verbose: bool = verbose
        builder = wp.sim.ModelBuilder()
        warp.sim.parse_urdf(
            URDF_PATH,
            builder,
            floating=False,
            xform=wp.transform([0.0, 0.05, 0.0], wp.quat_from_axis_angle(wp.vec3(1.0, 0.0, 0.0), -np.pi * 0.5)),
        )

        self.num_frames = num_frames
        self.frame_dt = 1.0 / fps
        self.substeps: int = substeps
        self.sim_dt = self.frame_dt / self.substeps
        self.sim_time: float = 0.

        self.model = builder.finalize()
        self.model.ground = True

        if integrator == 'featherstone':
            self.integrator = wp.sim.FeatherstoneIntegrator(self.model)
        elif integrator == 'semi-implicit':
            self.integrator = wp.sim.SemiImplicitIntegrator()
        else:
            raise ValueError(f"Unknown integrator: {integrator}")

        if render:
            self.renderer = wp.sim.render.SimRenderer(self.model, stage_path, scaling=1.0)

        self.state_0 = self.model.state()
        self.state_1 = self.model.state()

        if integrator == 'featherstone':
            wp.sim.eval_ik(self.model, self.state_0, self.state_0.joint_q, self.state_0.joint_qd)
        else:
            wp.sim.eval_fk(self.model, self.model.joint_q, self.model.joint_qd, None, self.state_0)

    def forward(self):
        self.render()
        for i in range(self.num_frames):
            with wp.ScopedTimer("forward step", active=self.verbose):
                for j in range(self.substeps):
                    self.state_0.clear_forces()
                    wp.sim.collide(self.model, self.state_0)
                    self.integrator.simulate(self.model, self.state_0, self.state_1, self.sim_dt)
                    self.state_0, self.state_1 = self.state_1, self.state_0
                    self.sim_time += self.sim_dt
                self.render()

    def render(self):
        """Renders the state history at self.fps fps to a USD."""
        with wp.ScopedTimer("render", active=self.verbose):
            self.renderer.begin_frame(self.sim_time)
            self.renderer.render(self.state_0)
            self.renderer.end_frame()

if __name__ == '__main__':
    wp.init()
    args = {'num_frames': 45,
            'integrator': 'semi-implicit',
            'substeps': 2000,
            'render': True,
            'verbose': True
            }
    sim = RobotSim(**args)
    sim.forward()
    sim.renderer.save()
```

[image]

[image]

[image]

eric-heiden, 2025-02-05:
Thanks for these suggestions, @JonathanKuelz! We will improve the documentation on this.
Just to clarify, the SemiImplicitIntegrator does support articulated mechanisms, it typically needs a lot more simulation substeps though than XPBD and Featherstone. You may also need to increase `model.joint_attach_ke` and `model.joint_attach_kd` by a lot (maybe even 100x), see `example_cartpole.py` which uses this integrator by default.

JonathanKuelz, 2025-02-05:
Thank you for the reply @eric-heiden  - even after ramping up both `joint_attach_ke` and `joint_attach_kd` I need to perform as many as 10000 substeps (compared to ~20 w. featherstone) to get a reasonable output. Just wanted to make sure this is the expected behavior.

closed 2025-11-06 by shi-eric