NVIDIA/warp #497: [DOCS] Document matrix column constructor

daedalus5, 2025-02-05:
## Category

- [ ] Report an error in the documentation.
- [x] Request for something to be documented.
- [ ] Suggestion to improve the documentation.
- [ ] Other (please explain)

## Description

The column-wise constructor for matrices needs to be documented, as it isn't clear to all users that rows shouldn't be used instead.

shi-eric, 2025-02-05:
Also note the related issue GH-278

christophercrouzet, 2025-02-24:
Closing due to not being necessary anymore after deprecating the constructor `wp.matrix()` from accepting vectors, and the introduction of the `wp.matrix_from_cols()` and `wp.matrix_from_rows()` built-ins in commit [1645944](https://github.com/NVIDIA/warp/commit/16459440de446f770f0b299c57ef04573ab4dec6).

closed 2025-02-24 by christophercrouzet