NVIDIA/warp #531: [BUG] Fix deprecation warnings with Python 3.12

mmacklin, 2025-02-19:
### Bug Description

The following warnings are encountered with Python 3.12:

```
env/lib/python3.12/site-packages/warp/codegen.py:2720
  /usr/local/google/home/erikfrey/proj/mjx_warp/env/lib/python3.12/site-packages/warp/codegen.py:2720: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
    ast.Str: emit_String,  # Deprecated in 3.8; use Constant
```

shi-eric, 2025-02-19:
Here's one way to see deprecation warnings from `ast`:

```python
import warnings

import warp as wp

# Enable all DeprecationWarnings (including those from other libraries)
warnings.filterwarnings("default", category=DeprecationWarning)

wp.clear_kernel_cache()

@wp.kernel
def test_print_kernel():
    wp.printf("this is a float %f\n", 457.5)

wp.load_module(device="cpu")
```

Output:

```
Warp 1.6.0 initialized:
   CUDA Toolkit 12.8, Driver 12.8
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 3090" (24 GiB, sm_86, mempool enabled)
     "cuda:1"   : "NVIDIA GeForce RTX 3090" (24 GiB, sm_86, mempool enabled)
   CUDA peer access:
     Supported fully (all-directional)
   Kernel cache:
     /home/eshi/.cache/warp/1.6.0
/home/eshi/code-projects/warp/warp/codegen.py:1886: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
  if isinstance(node, ast.Str):
/home/eshi/code-projects/warp/warp/codegen.py:1849: DeprecationWarning: Attribute s is deprecated and will be removed in Python 3.14; use value instead
  return adj.add_constant(node.s)
/home/eshi/code-projects/warp/warp/codegen.py:1888: DeprecationWarning: ast.Num is deprecated and will be removed in Python 3.14; use ast.Constant instead
  elif isinstance(node, ast.Num):
/home/eshi/code-projects/warp/warp/codegen.py:1853: DeprecationWarning: Attribute n is deprecated and will be removed in Python 3.14; use value instead
  key = (node.n, type(node.n))
/home/eshi/code-projects/warp/warp/codegen.py:1858: DeprecationWarning: Attribute n is deprecated and will be removed in Python 3.14; use value instead
  out = adj.add_constant(node.n)
Module __main__ b111dd7 load on device 'cpu' took 1312.08 ms  (compiled)
```

shi-eric, 2025-02-27:
Should be fixed by 92fabfdf496ac6178f2cd793c8beaf5bf08f6b06

closed 2025-02-27 by shi-eric