NVIDIA/warp #534: [QUESTION] Gradients wrt. to vertex position in FEM simulation

andrinr, 2025-02-20:
Can we compute the gradients with respect to the mesh vertex position in a finite element simulation. If yes are there any examples that do so?

shi-eric, 2025-03-03:
Are you asking about `warp.fem` specifically?

andrinr, 2025-03-04:
yes

gdaviet, 2025-03-05:
Hi @andrinr, I think the short answer is yes -- I am not sure we have an existing example that does exactly that, but happy to provide one. I would need a bit more context to answer more precisely.
Do you mean:
 1. Computing the gradient of a `fem.integrate` or `fem.interpolate` call with respect to the vertices of the underlying geometry? If so, that is possible using the usual backward (adjoint) differentiation: capture the integrate call in a `wp.Tape` and make sure the underlying geometry vertex position array has `requires_grad=True`, e.g, if `geo` is a `fem.Tetmesh` or `fem.Hexmesh`, `geo.positions.requires_grad=True`. Note that bilinear forms (sparse matrices) are not currently differentiable, though that is something I want to fix
 2. Computing the gradient of a field or expression with respect to a space degrees of freedom (which will coincide with the mesh vertices for linear interpolation). In the current dev version of warp that is also possible in a forward differentiation way (i.e. computing the full gradient rather than the adjoint w.r.t some loss), by passing a "trial" field to `fem.interpolate`. The result will be a sparse matrix containing the gradient of the interpolated function with respect to trial field degrees of freedom at each quadrature epoint

andrinr, 2025-03-05:
Hi @gdaviet :) I am mostly interested in the first case. If you could come up with an example, that would be wonderful. 
If it helps, I have some ideas:

- Computing the gradients of the boundary nodes of a (2d) cylinder in a RANS airflow simulation. The objective function would be the mean of the RANS field. If that works, a shape optimization could be tried.
- The same thing as above, but instead of coming up with a RANS sim, it could be a simple adapation of the[ 2D NS ](https://github.com/NVIDIA/warp/blob/main/warp/examples/fem/example_navier_stokes.py) example. The objective would be minimzing the Velocity Fluctuations[ (Turbulent Kinetic Energy - TKE)](https://en.wikipedia.org/wiki/Turbulence_kinetic_energy).

gdaviet, 2025-03-05:
Cool, I'll try to setup a shape optimization example, I agree that it would be good to have one. Probably with a linear model to keep it simple