NVIDIA/warp #544: [BUG] Cannot visualize the example output usd files,

Tianyi20, 2025-02-24:
### Bug Description

Hi, just try several core examplese and generate several usd files. But I encounter the problem when I use usd viewer and omniverse to visualize. Blender can visualize but I cannot see the live stream.

I tried the three core examples

[image]

This is the usd viewer error info.
```
(usd) tianyi@mei-MS-7B48:~/usd_new/scripts$ ./usdview_gui.sh
Activated USD python/tools from /home/tianyi/usd_new
Traceback (most recent call last):
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 2923, in _reopenStage
    stage = self._openStage(
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 1260, in _openStage
    stage = Usd.Stage.Open(layer,
pxr.Tf.ErrorException: 
	Error in 'pxrInternal_v0_23__pxrReserved__::UsdStage::Open' at line 1102 in file /buildAgent/work/4d6ec4296c0480b4/USD/pxr/usd/usd/stage.cpp : 'Invalid root layer'
Traceback (most recent call last):
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 3500, in _primSelectionChanged
    self._updatePrimViewSelection(added, removed)
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 3617, in _updatePrimViewSelection
    self._ui.primView.updateSelection(addedItems, removedItems)
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/primTreeWidget.py", line 486, in updateSelection
    self._refreshAncestorsOfSelected()
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/primTreeWidget.py", line 502, in _refreshAncestorsOfSelected
    for prim in self._appController._dataModel.selection.getPrims()]
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/selectionDataModel.py", line 724, in getPrims
    return [self._rootDataModel.stage.GetPrimAtPath(path)
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/selectionDataModel.py", line 724, in <listcomp>
    return [self._rootDataModel.stage.GetPrimAtPath(path)
AttributeError: 'NoneType' object has no attribute 'GetPrimAtPath'
Traceback (most recent call last):
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/stageView.py", line 2357, in _primSelectionChanged
    self.updateSelection()
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/stageView.py", line 1365, in updateSelection
    psuRoot = self._dataModel.stage.GetPseudoRoot()
AttributeError: 'NoneType' object has no attribute 'GetPseudoRoot'
Traceback (most recent call last):
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 3106, in _propSelectionChanged
    self._updatePropertyViewSelection()
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 3823, in _updatePropertyViewSelection
    focusPrim = self._dataModel.selection.getFocusPrim()
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/selectionDataModel.py", line 719, in getFocusPrim
    return self._rootDataModel.stage.GetPrimAtPath(self.getFocusPrimPath())
AttributeError: 'NoneType' object has no attribute 'GetPrimAtPath'
Traceback (most recent call last):
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 2923, in _reopenStage
    stage = self._openStage(
  File "/home/tianyi/usd_new/lib/python/pxr/Usdviewq/appController.py", line 1260, in _openStage
    stage = Usd.Stage.Open(layer,
pxr.Tf.ErrorException: 
	Error in 'pxrInternal_v0_23__pxrReserved__::UsdStage::Open' at line 1102 in file /buildAgent/work/4d6ec4296c0480b4/USD/pxr/usd/usd/stage.cpp : 'Invalid root layer'

```

I also tried to use omni to visualize

```
2025-02-24 05:24:15 [47,510ms] [Error] [omni.usd] Failed to open : /home/tianyi/usd/example_dem.usd
2025-02-24 05:24:15 [47,533ms] [Error] [omni.kit.window.file.file_window] Failed to open stage /home/tianyi/usd/example_dem.usd: Failed to open:/home/tianyi/usd/example_dem.usd
2025-02-24 05:24:33 [66,183ms] [Error] [omni.usd] Failed to open : /home/tianyi/usd/example_nvdb.usd
2025-02-24 05:24:33 [66,203ms] [Error] [omni.kit.window.file.file_window] Failed to open stage /home/tianyi/usd/example_nvdb.usd: Failed to open:/home/tianyi/usd/example_nvdb.usd

```

Blender can help to visualize the initial scene but fail to have live stream video,

[image]

shi-eric, 2025-02-24:
I'm not sure this is the issue you're running into, but it feels similar to the breaking changes for USDC file formats in USD: https://github.com/NVIDIA/warp/issues/462#issuecomment-2617243783

Assuming you're on version 25.2 of usd-core, you can also try downgrading to 24.11.

eric-heiden, 2025-02-24:
Another option (although slower and takes more space) is to write to a USDA file (rather than the binary USD format) by setting `.usda` as the file extension in the USD renderer.

zuodexin, 2025-03-08:
same problem

shi-eric, 2025-03-10:
> same problem

Okay, what have you tried?

grindstm, 2025-03-17:
I have the same problem and would like to use omniverse for display. Attempting to load these examples in omniverse gives `Failed to open stage...`. 
The stages can be loaded and animated using usdview instead. In KDE Plasma under Wayland, to use GPU, I must:
```
export QT_QPA_PLATFORM=xcb
prime-run usdview
```
usdview then throws a steady stream of `ERROR: Usdview encountered an error while rendering.Attempt to retrieve context when no valid context`, but it does display. Note that dome_light needs to be deactivated to see anything. 

```
Warp 1.6.2 initialized:
   CUDA Toolkit 12.8, Driver 12.8
   Devices:
     "cpu"      : "CPU"
     "cuda:0"   : "NVIDIA GeForce RTX 3060" (12 GiB, sm_86, mempool enabled)
```