NVIDIA/warp #547: [BUG] jax_callable and wp.to_jax() can't co-exist

HaoliangWang, 2025-02-24:
### Bug Description

I'm following the document [here](https://github.com/NVIDIA/warp/blob/5c47ddd5853b36d033325f8c0d26b15b5d9ad7d9/docs/modules/interoperability.rst#calling-annotated-python-functions) to learn about the new `jax_callable` feature, the code snippet in the document runs without errors.

In my application, I need to use `jax_callable` and `wp.to_jax()` together. However, when I add `wp.to_jax()` to the example code snippet, it crashes. See minimal example below:

> import jax
> import jax.numpy as jnp
> import warp as wp
> from warp.jax_experimental.ffi import jax_callable
> 
> 
> @wp.kernel
> def scale_kernel(a: wp.array(dtype=float), s: float, output: wp.array(dtype=float)):
>     tid = wp.tid()
>     output[tid] = a[tid] * s
> 
> @wp.kernel
> def scale_vec_kernel(a: wp.array(dtype=wp.vec2), s: float, output: wp.array(dtype=wp.vec2)):
>     tid = wp.tid()
>     output[tid] = a[tid] * s
> 
> 
> def example_func(
>     # inputs
>     a: wp.array(dtype=float),
>     b: wp.array(dtype=wp.vec2),
>     s: float,
>     # outputs
>     c: wp.array(dtype=float),
>     d: wp.array(dtype=wp.vec2),
> ):
>     wp.launch(scale_kernel, dim=a.shape, inputs=[a, s], outputs=[c])
>     wp.launch(scale_vec_kernel, dim=b.shape, inputs=[b, s], outputs=[d])
> 
> 
> jax_func = jax_callable(example_func, num_outputs=2)
> 
> a = wp.array([1, 2, 3])
> b = wp.to_jax(a)

The error message is:

> Traceback (most recent call last):
>   File "test.py", line 31, in <module>
>     b = wp.to_jax(a)
>         ^^^^^^^^^^^^
>   File "/python3.11/site-packages/warp/jax.py", line 164, in to_jax
>     return jax.dlpack.from_dlpack(warp.to_dlpack(warp_array))
>                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/python3.11/site-packages/warp/dlpack.py", line 297, in to_dlpack
>     capsule = PyCapsule_New(
>               ^^^^^^^^^^^^^^
> ctypes.ArgumentError: argument 3: TypeError: expected CFunctionType instance instead of CFunctionType
> 

If I comment out the line `jax_func = jax_callable(example_func, num_outputs=2)`, the error goes away, but I need both `jax_callable` and `wp.to_jax()`, how should I fix this? Thanks!

### System Information

I'm using warp `1.7.0.dev20250223`, python `3.11`.

nvlukasz, 2025-02-24:
Hi Haoliang, thanks for the repro. I see the same error and I'll try to figure out what's happening there.

nvlukasz, 2025-02-25:
Fixed in 6ab64ce7b3ece6596b590e01806bb2bde7a87cc4

closed 2025-02-25 by nvlukasz