NVIDIA/warp #553: [BUG] jax_callable functions ignores initialized output values for wp.kernel

HaoliangWang, 2025-02-26:
### Bug Description

I want to initialize the output value before launching wp.kernel in a jax_callable function, and let the code in the kernel modify the value. 

For example, in regular warp code, I can do the following:

> import warp as wp
> 
> @wp.kernel
> def scale_kernel(a: wp.array(dtype=int), b: wp.array(dtype=int)):
>     tid = wp.tid()
>     wp.atomic_add(b, 0, a[tid])
> 
> 
> def example_func():
>     a = wp.array([1, 2, 3], dtype=wp.types.int32)
>     b = wp.array([10], dtype=wp.types.int32)
>     wp.launch(scale_kernel, dim=a.shape, inputs=[a], outputs=[b])
>     print(b)
> 
> example_func()

and this gives the expected results which is `[16]`. However, if I make `example_func` a jax primitive by doing the following:

> from warp.jax_experimental.ffi import jax_callable
> import warp as wp
> import jax
> 
> @wp.kernel
> def scale_kernel(a: wp.array(dtype=int), b: wp.array(dtype=int)):
>     tid = wp.tid()
>     wp.atomic_add(b, 0, a[tid])
> 
> 
> def example_func(
>     # outputs
>     b: wp.array(dtype=int)
> ):
>     a = wp.array([1, 2, 3], dtype=wp.types.int32)
>     b = wp.array([10], dtype=wp.types.int32)
>     wp.launch(scale_kernel, dim=a.shape, inputs=[a], outputs=[b])
> 
> jax_func = jax_callable(example_func, num_outputs=1)
> 
> @jax.jit
> def f():
>     # output shapes
>     output_dims = {"b": 1}
>     b = jax_func(output_dims=output_dims)
>     return b
> 
> r = f()
> print(r)

it seems that the wp.kernel just ignores the initialized output value (`b = wp.array([10], dtype=wp.types.int32)`) and gives `[Array([0], dtype=int32)]` as results.

How do I use `wp.kernel` to modify values (`b` in this case) outside of it in a `jax_callable` function? I'm not sure if this is a bug or should it be a feature request. Thanks for your help!

### System Information

warp version `1.7.0.dev20250223`.

nvlukasz, 2025-02-26:
Btw, having arrays that are used for both input and output is a little tricky and I'm currently trying to improve the support. The challenge is that the output arrays are allocated by JAX prior to the call, so JAX is in charge of initializing them. Also JAX generally doesn't like in-place modifications of arrays, whereas Warp supports it just fine. I'm playing with the `input_output_aliases` argument to JAX's `ffi_call()` to see if it can help to can support in-place modifications with `jax_callable()` and `jax_kernel()`.

HaoliangWang, 2025-02-26:
Thanks for the reply! Yes that sounds right, what I'm really trying to achieve here is to make the wp.kernel modify an array defined outside the kernel (and that's why I gave `b` an _initial_ value of 10 and see if the kernel can increment it to 16). I've also tried passing an array ptr as an argument to the kernel mentioned in [this issue](https://github.com/NVIDIA/warp/issues/206), but it seems that it doesn't work with `jax_callable` either.

```
from warp.jax_experimental.ffi import jax_callable
import warp as wp
import jax

@wp.kernel
def scale_kernel(a: int, ptr: wp.uint64, b: wp.array(dtype=int)):
    b = wp.array(ptr=ptr, shape=(1,), dtype=int)
    wp.atomic_add(b, 0, a)

def example_func(
    # outputs
    b: wp.array(dtype=int)
):
    a = 3
    b = wp.array([10], dtype=int)
    wp.launch(scale_kernel, dim=(1,), inputs=[a, b.ptr], outputs=[b])

jax_func = jax_callable(example_func, num_outputs=1)

@jax.jit
def f():
    # output shapes
    output_dims = {"b": 1}
    b = jax_func(output_dims=output_dims)
    return b

r= f()
print(r)
```

The output is still `[Array([0], dtype=int32)]`.

nvlukasz, 2025-02-26:
Right, the reason you get 0 in the output is that you're not writing to the actual output array that was prepared by JAX.

To do what you want, you need to pass the initial value of `b` as an input and write the final value of `b` as an output.  Here's a working example:

```python
from warp.jax_experimental.ffi import jax_callable
import warp as wp
import jax
import jax.numpy as jnp

@wp.kernel
def scale_kernel(a: wp.array(dtype=int), b: wp.array(dtype=int)):
    tid = wp.tid()
    wp.atomic_add(b, 0, a[tid])

def example_func(
        b_in: wp.array(dtype=int),
        # outputs
        b_out: wp.array(dtype=int),
    ):
    a = wp.array([1, 2, 3], dtype=wp.types.int32)

    # copy the initial values into the output array
    wp.copy(b_out, b_in)
    
    # write to the output array
    wp.launch(scale_kernel, dim=a.shape, inputs=[a], outputs=[b_out])

jax_func = jax_callable(example_func, num_outputs=1)

@jax.jit
def f():
    # output shapes
    b = jnp.array([10], dtype=jnp.int32)
    b = jax_func(b)
    return b

r = f()
print(r)
```

nvlukasz, 2025-02-26:
Here's an alternative solution that doesn't require adding an input array. Instead, we initialize the output array in the callable:

```python
from warp.jax_experimental.ffi import jax_callable
import warp as wp
import jax
import jax.numpy as jnp

@wp.kernel
def scale_kernel(a: wp.array(dtype=int), b: wp.array(dtype=int)):
    tid = wp.tid()
    wp.atomic_add(b, 0, a[tid])

def example_func(
        # outputs
        b: wp.array(dtype=int),
    ):
    a = wp.array([1, 2, 3], dtype=wp.types.int32)

    # initialize the output array
    b.fill_(10)
    
    # write to the output array
    wp.launch(scale_kernel, dim=a.shape, inputs=[a], outputs=[b])

jax_func = jax_callable(example_func, num_outputs=1)

@jax.jit
def f():
    # output shapes
    output_dims = {"b": 1}
    b = jax_func(output_dims=output_dims)
    return b

r = f()
print(r)
```

HaoliangWang, 2025-02-26:
Thanks! Both approaches work for my purposes, this is extremely helpful! Closing this issue for now.

closed 2025-02-26 by HaoliangWang