NVIDIA/warp #557: [BUG] multiple cloth grids in scene

etaoxing, 2025-02-28:
### Bug Description

I'm upgrading [Rewarped](https://rewarped.github.io/) from `warp-lang==1.3.3` to the nightly build `warp-lang>1.6`. Seems like I'm no longer able to spawn multiple cloth grids in the same scene without the simulation blowing up.

```python
def create_cloth(self, builder):
      builder.add_cloth_grid(
          pos=wp.vec3(0.0, 0.0, 0.0),
          vel=wp.vec3(0.1, 0.1, 0.0),
          rot=wp.quat_from_axis_angle(wp.vec3(1.0, 0.0, 0.0), -math.pi * 0.25),
          dim_x=dim_x,
          dim_y=dim_y,
          cell_x=1.0 / dim_x,
          cell_y=1.0 / dim_y,
          mass=1.0,
          tri_ke=10000.0,
          tri_ka=10000.0,
          tri_kd=100.0,
          tri_lift=10.0,
          tri_drag=5.0,
      )

  def create_builder(self):
      builder = self.create_modelbuilder()
      env_builder = self.create_modelbuilder()
      self.create_cloth(env_builder)
      self.env_offsets = compute_env_offsets(self.num_envs, self.env_offset, self.env_offset_correction, self.up_axis)
      for i in range(self.num_envs):
          xform = wp.transform(self.env_offsets[i], wp.quat_identity())
          builder.add_builder(
              env_builder,
              xform,
              update_num_env_count=True,
              separate_collision_group=self.separate_collision_group_per_env,
          )
      return builder
```

Based on `warp.examples.optim.cloth_throw_example`.

Multiple instances of `warp.examples.optim.walker` still works with `builder.add_soft_mesh(..)`

### System Information

on warp commit https://github.com/NVIDIA/warp/commit/92fabfdf496ac6178f2cd793c8beaf5bf08f6b06

Python 3.10, Ubuntu 22.04, cuda 12.5

shi-eric, 2025-02-28:
Is there a script you can provide to generate `ClothThrowWarpExamples_2.usd` so we can use it to bisect the commits?

etaoxing, 2025-03-01:
It's [here](https://github.com/rewarped/rewarped/blob/v1.3.0/rewarped/envs/warp_examples/cloth_throw.py). 

```bash
pip install rewarped
python -m rewarped.envs.warp_examples.cloth_throw --num_envs 2
```

Did some more digging though and it seems like an issue on my end. Works with the warp nightly version and some older code that I have for Rewarped, while the link I shared is broken on `warp-lang==1.3.3` and `warp-lang>1.6`.

etaoxing, 2025-03-01:
That old code was using a [`ModelBuilder`](https://github.com/NVIDIA/warp/blob/v1.1.0/warp/sim/model.py) from `warp-lang==1.1.0`.

When I run `python -m rewarped.envs.warp_examples.cloth_throw --num_envs 2` with `ModelBuilder` from `1.3.3`, the simulation blows up. Whereas when I swap in a `ModelBuilder` from `1.1.0`, the example works correctly.

 So it seems like this is definitely a regression introduced between `1.1.0` to `1.3.3`. 

[model_110.py.txt](https://github.com/user-attachments/files/19040858/model_110.py.txt)

AnkaChan, 2025-03-12:
Hi Etao can you prepare a minimalist demo that reproduces the error? Currently, We have to step into your repo to locate the error which is quite time-consuming considering how complicated your repo is. We need you to help us to narrow down the error to our side.

etaoxing, 2025-03-12:
[multi_cloth_throw.py.txt](https://github.com/user-attachments/files/19218291/multi_cloth_throw.py.txt)

[cloth_outputs.zip](https://github.com/user-attachments/files/19218292/cloth_outputs.zip)

AnkaChan, 2025-03-13:
After the update in 1.4.0, edges can have a vertex index of -1 to denote an open edge. When merging a new builder with an old one, we perform the following steps:

> self.edge_indices.extend((np.array(builder.edge_indices, dtype=np.int32) + start_particle_idx).tolist())

which can change those -1s to positive values and create crazy edges.
I am on it.

shi-eric, 2025-03-14:
@AnkaChan's fix went in with bfad03bc7421ec9405f77fbd0a303a8411dd61ba and will be in the nightly made tonight, please give it a try when you have a chance.

etaoxing, 2025-03-14:
The fix works! Thanks Anka and Eric!

[ClothThrowWarpExamples_4.usd.zip](https://github.com/user-attachments/files/19247837/ClothThrowWarpExamples_4.usd.zip)

closed 2025-03-14 by etaoxing