NVIDIA/warp #562: [QUESTION] How to give rotation to warp's openGL camera

Tianyi20, 2025-03-04:
Hi, I am trying to give rotation to the camera. But I tried many times, It seems the camera_front dosn't control camera's pointing rotation? How can I rotate the camera try to make it inclined to see the scene?

The camera always pointing straightly to the scene. Do you know how to apply rotation?

[image]

```
# Initialize a hash grid.
grid = wp.HashGrid(128, 128, 128)

# Cell size to use when building the grid.
grid_cell_size = point_radius * 5.0

# Camera settings.
#camera_pos = (-26.0, 6.0, 12.5)
#camera_pos = (-26.0, 6.0, 12.5)
camera_pos = (-20.0, 5.0, 5.5)

camera_front = (0.707, 0, 0.707)

# Create a headless OpenGL renderer for our scene.
renderer = warp.render.OpenGLRenderer(
    fps=fps,
    screen_width=resolution[0],
    screen_height=resolution[1],
    #camera_fov=45.0,
    camera_pos=camera_pos,
    camera_front=camera_front,
    camera_up=(0.0, 1.0, 0.0),
    draw_grid=False,
    draw_axis=False,
    vsync=False,
    headless=False,
)
```

christophercrouzet, 2025-03-25:
Hi @Tianyi20, apologies for the late reply!

You could try something like the attached snippet where you compute the `camera_front` value based on the camera's position and target.

```python
import numpy as np

import warp as wp
import warp.render

BOX_POINTS = np.array(
    (
        ( 0.853553, -0.146446,  0.0     ), ( 0.146446, -0.853553,  0.0     ),
        ( 0.353553,  0.353553,  0.707106), (-0.353553, -0.353553,  0.707106),
        (-0.353553, -0.353553, -0.707106), ( 0.353553,  0.353553, -0.707106),
        (-0.853553,  0.146446,  0.0     ), (-0.146446,  0.853553,  0.0     ),
    ),
    dtype=np.float32,
)

BOX_FACE_VERTEX_INDICES = np.array(
    (
        0, 3, 1, 0, 2, 3, 4, 7, 5, 4, 6, 7, 6, 2, 7, 6, 3, 2,
        5, 1, 4, 5, 0, 1, 5, 2, 0, 5, 7, 2, 1, 6, 4, 1, 3, 6,
    ),
    dtype=np.int32,
)

resolution = (512, 384)
num_frames = 240
fps = 24

camera_pos = wp.vec3(0.0, 2.5, 5.0)
camera_target = wp.vec3(0.0, 0.0, 0.0)
camera_front = wp.normalize(camera_target - camera_pos)

renderer = warp.render.OpenGLRenderer(
    fps=fps,
    screen_width=resolution[0],
    screen_height=resolution[1],
    camera_pos=camera_pos,
    camera_front=camera_front,
    vsync=True,
)

for frame in range(num_frames):
    renderer.begin_frame(frame / num_frames)
    renderer.render_mesh(
        "box",
        BOX_POINTS,
        BOX_FACE_VERTEX_INDICES,
        pos=(0.0, 0.0, 0.0),
        colors=(0.25, 0.5, 0.65),
    )
    renderer.end_frame()
```

christophercrouzet, 2025-03-25:
I'll be closing this issue now but please feel free to reopen it if that's not what you were looking for!

closed 2025-03-25 by christophercrouzet