NVIDIA/warp #566: [QUESTION] Hard SDF contacts

JonathanKuelz, 2025-03-06:
Dear warp team,

browsing the examples and the source code, it seems like warp is currently only supporting `SDF` collisions in soft body simulation. Is this assumption correct or is there a way to define rigid geometries using signed distance fields and make them collide?

hejia-zhang, 2025-03-14:
Do you also support computing the distance between two SDFs?

eric-heiden, 2025-04-01:
We currently don't have rigid collisions between SDF in Warp.

To extract the zero level set from a SDF you can use our `wp.MarchingCubes` implementation, see [example_marching_cubes.py](https://github.com/NVIDIA/warp/blob/main/warp/examples/core/example_marching_cubes.py).