NVIDIA/warp #577: [QUESTION]  Use Franka arm to hit a box. How to right the code?

HaotianHehaha, 2025-03-17:
As the title says, I add a Franka arm and try to control the arm to hit a box on the ground. But I am confused how to write the code. In the definition of forward function, should I use `warp.sim.eval_fk()` or  `warp.sim.collide()` to detect the collision points and use an integrator (like 'warp.sim.FeatherstoneIntegrator') to simulate the collision? Or use them together?

Whether the integrator in Warp include simulation like forward kinematics?

AnkaChan, 2025-03-17:
Hi HotianHehaha, first of all, how do you represent your box? Is it a rigid body or soft body?

Either way, FeatherStone can simulate them all, because FeatureStoneIntegrator supports both rigid body and soft body simulation. After adding the box and the robot to model, you can initialize a `FeatureStoneIntegrator` (for example as 'integrator'). You need to call 'wp.sim.collide' to detect the collision and call `integrator.simulate` to advance the simulator.

As to controlling the robot, you can either explicitly edit it's translation and rotation in its `state`, or modify the force in  `state` to control its dynamics. You can also use IK to calculate the desired torque, see this example: https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_jacobian_ik.py

HaotianHehaha, 2025-03-18:
I got it. Thanks for your suggestions!

closed 2025-03-18 by shi-eric