NVIDIA/warp #580: [QUESTION] What's the relationship of Warp and cuTile?

chaoming0625, 2025-03-19:
Recently, NVIDIA announced cuTile, a tile programming model for CUDA. It introduces an array-based paradigm where the compiler automatically handles memory movement, pipelining, and tensor core utilization, making GPU programming more accessible and portable. I'm curious about the relationship between cuTile and Warp. In the future, will Warp be based on cuTile?

jc211, 2025-03-20:
I can't find any documentation about this other than this post: https://x.com/blelbach/status/1902113767066103949

mmacklin, 2025-03-22:
The tile based programming model in Warp is very similar to what will be available in cuTile so it's likely we'll be able to use it for some of our operations in the future. In the meantime Warp will continue to use other kernel level libraries like cuBLASDx, cuSOLVERDx for our accelerated tile ops.

closed 2025-03-22 by mmacklin

jsonlee0x02, 2025-04-03:
cuTile is more like OpenAI's [triton](https://github.com/triton-lang/triton)