NVIDIA/warp #588: [BUG] Non-determinism with VDB and self contact resolution

CliffieVanR, 2025-03-20:
### Bug Description

We've been noticing some non-deterministic behaviour using the VDB integrator for cloth simulation when self contact handling is enabled. I believe the cause is lies here in vertex_triangle_collision_detection_kernel inside collide.py:

            min_dis_to_tris = wp.min(min_dis_to_tris, dist)
            if vertex_num_collisions < vertex_buffer_size:
                vertex_colliding_triangles[vertex_buffer_offset + vertex_num_collisions] = tri_index
            else:
                resize_flags[VERTEX_COLLISION_BUFFER_OVERFLOW_INDEX] = 1

            vertex_num_collisions = vertex_num_collisions + 1

Note that the number of collisions is increased even if the allotted space is full. I believe this causes the code that resolves the collisions to read beyond a particle's own collisions and from its neighbours. You'll like need a complex collision mesh to reproduce the problem.

AnkaChan, 2025-03-20:
Hi Cliffie please see those functions in `wp.collide` in line 2077~2092:

> @wp.func
> def get_vertex_colliding_triangles_count(col_info: TriMeshCollisionInfo, v: int):
>     return wp.min(col_info.vertex_colliding_triangles_count[v], col_info.vertex_colliding_triangles_buffer_sizes[v])
> 
> 
> @wp.func
> def get_vertex_colliding_triangles(col_info: TriMeshCollisionInfo, v: int, i_collision: int):
>     offset = col_info.vertex_colliding_triangles_offsets[v]
>     return col_info.vertex_colliding_triangles[offset + i_collision]
> 
> 
> @wp.func
> def get_triangle_colliding_vertices_count(col_info: TriMeshCollisionInfo, tri: int):
>     return wp.min(
>         col_info.triangle_colliding_vertices_count[tri], col_info.triangle_colliding_vertices_buffer_sizes[tri]
>     )

Even though `vertex_colliding_triangles_count` can be larger than `vertex_colliding_triangles_buffer_sizes` we will not read beyond the allocated space.

The true source of non-determinism is from here:

>             tri_num_collisions = wp.atomic_add(triangle_colliding_vertices_count, tri_index, 1)
see line 1909 of `wp.collide` .

Multi-thread will access `triangle_colliding_vertices` in random order due to the atomic operation. If you truly want determinism you can sort `triangle_colliding_vertices`  after the collision detection. We do not do this because we do not claim to be a deterministic simulator.

CliffieVanR, 2025-03-21:
Hi Anka,

Many thanks for the quick response. That makes total sense.

Best,
Cliff

closed 2025-03-21 by shi-eric