NVIDIA/warp #589: [QUESTION] Questions on warp.fem elasticity examples

akrivx, 2025-03-20:
Hi,

I have a few questions regarding the elasticity examples in `warp.fem`. They may be basic, but I'd appreciate any guidance.

1. In both the **mixed elasticity** and **nonconforming contact** examples, stress fields are used as degrees of freedom. What would the corresponding **linear and bilinear forms** look like for a **displacement-based elasticity formulation** (i.e., without stress DOFs)?
   
2. In **displacement-based elasticity**, how can I write a **linear form** that directly computes the **internal force vector** from an already evaluated displacement field? That is, instead of forming the **stiffness matrix** and multiplying it by displacements, I’d like to obtain the internal forces **directly** via an integral.

3. Are there any plans to add examples on **corotational FEM**? I'm particularly interested in incorporating **element rotation matrices** into the **linear and bilinear forms**.

Thanks,  
Chris

gdaviet, 2025-03-21:
Hi Chris,

Currently we do not ship an example with displacement-only fem in the core distribution, but please see the branch below for an (unfortunately more complex) example of how it is possible to write one. 
In `softbody_sim.py` the `nh_xxx` forms implement a Stable Neo--Hookean matetrial model, the `cr_xxx` a co-rotated one. 

https://github.com/gdaviet/warp/blob/a2775ea6db02a3c3ab1b5aa5707a82de12c44955/warp/examples/fem/example_cutting/softbody_sim.py#L752

akrivx, 2025-03-21:
Hi @gdaviet ,

Thanks for pointing me to the example—I’ve been studying the `cr_elastic_*` forms (especially `cr_elastic_forces`) and adapting them for a simple _explicit_ corotated dynamics setup.

Since the example is built around a quasi-static Newton solver, I wanted to ask:
Is it appropriate to reuse the `cr_elastic_forces` integrand directly in explicit dynamics, without forming the Hessian?

Also, I haven’t been able to run the example itself because I’m on an RTX 5080 (SM 12.0), so I’d need to build PyTorch from source. I’ve been relying on code inspection and partial tests for now.

Thanks again,
Chris

shi-eric, 2025-03-22:
> Hi [@gdaviet](https://github.com/gdaviet) ,
>
> Thanks for pointing me to the example—I’ve been studying the `cr_elastic_*` forms (especially `cr_elastic_forces`) and adapting them for a simple [...]

I think the Pytorch nightlies are now built with CUDA 12.8 and should support Blackwell. Congrats on the GPU by the way!

gdaviet, 2025-03-24:
Regarding the use in an explicit solver -- yes, you can definitely use `cr_elastic_forces` in a standalone manner to compute the elastic forces, without assembling the hessian. 
Regarding the torch dependency, it is only used for the dynamic mesh topology part of this example which is probably not relevant for you, here's a simpler torch-free launcher script: https://github.com/gdaviet/warp/blob/example_cutting/warp/examples/fem/example_cutting/cantilever.py

akrivx, 2025-03-24:
That's awesome, thank you. It works nicely and is also fast, with a frame rate of ~15 ms/frame using the default parameters.

If I understand it correctly, the example controls the Newton step using line search with an energy-based acceptance criterion, and thus the energy integrands and the additional `du` field.

gdaviet, 2025-03-24:
> If I understand it correctly, the example controls the Newton step using line search with an energy-based acceptance criterion, and thus the energy integrands and the additional `du` field.

Yes, exactly!

akrivx, 2025-04-15:
> Hi Chris,
>
> Currently we do not ship an example with displacement-only fem in the core distribution, but please see the branch below for an (unfortunately more complex) example of how it is possible [...]

Out of curiosity, do these elasticity models work with 3D triangular meshes or just tetrahedral ones?

gdaviet, 2025-04-15:
> Out of curiosity, do these elasticity models work with 3D triangular meshes or just tetrahedral ones?

Those currently assume a 3x3 deformation gradient (volumetric mesh, so tet, hex, grid, etc), but I suppose they could be adapted to model a 2d membrane energy

akrivx, 2025-04-22:
I’ve now tested both the model from the branch above and my own implementation based on Green strain with analytical SVD-based rotation derivatives. The setup consists of a flat rectangular mesh representing a garment, subjected to gravity.

Both corotational elasticity models behave correctly on hexahedral meshes, regardless of the mesh's orientation or the direction of gravity. They also work as expected on triangle meshes when the mesh lies in the XY plane and gravity acts in-plane (i.e., along the X or Y axis). However, when gravity acts out-of-plane (e.g., along Z while the mesh lies in XY), neither model produces the expected behaviour—the mesh remains still.

This behaviour is consistent across both implementations, so I suspect it may be related to a limitation in how `warp.fem` handles gradients or test/trial spaces for triangle elements, but I might be wrong.

I’ve posted a related question here: https://github.com/NVIDIA/warp/issues/671