NVIDIA/warp #596: [BUG] Pull effect for rigid body collisions

JonathanKuelz, 2025-03-24:
### Bug Description

Warp avoids (heavy) penetration in rigid body collisions by introducing a `rigid_contact_margin` parameter that seems to determine for which distance prior to collision contact forces are going to be computed.

My expectation would be that these forces act as a _soft_ relaxation of the actual contact forces (i.e., the forces that act when the bodies actually are in contact). This would mean that the effect of the forces "pushes" the objects in collision away from each other.

Instead, in simulation, I have observed an effect of "contact pulls": As soon as two bodies are within the contact margin, they experience (extremely high) forces pulling them towards each other. The effect can be seen in the attached video.

https://github.com/user-attachments/assets/98767507-2379-48f7-979a-fc158ea46828

This is also reflected by the body forces in the integrated state. Here's a history of forces during the critical frames for the simulation shown in the video:

```
1) [   21.394   383.681  -857.057 -4807.015   479.997   328.227]
2) [   47.065   198.962  -899.292   649.488  1121.078   153.875]
3) [   16.723   336.591  -648.491   -24.052   606.769   501.74 ]
4) [  11.003  210.688 -437.894  154.427  109.905  209.24 ]
```

In the simulation, the only force that is applied externally is a control force along the x-direction of magnitude 50.

AnkaChan, 2025-03-24:
Hi would you mind providing us the version of warp you are using and the type of integrator you are using?

eric-heiden, 2025-04-01:
Hi @JonathanKuelz, would you be able to provide this experiment code so we can reproduce this?

JonathanKuelz, 2025-04-01:
Hmm, I upgraded to warp 1.7 today and now I can't reproduce the issue. I am going to close this and re-open in case I come across something similar again.

closed 2025-04-01 by JonathanKuelz