NVIDIA/warp #599: [BUG] module 'warp' has no attribute 'get_mempool_used_mem_current'

OmidGhotbi, 2025-03-26:
### Bug Description

python example_fluid_checkpoint.py
Warp 1.6.2 initialized:
   CUDA Toolkit 12.8, Driver 12.7
   Devices:
     "cpu"      : "Intel64 Family 6 Model 183 Stepping 1, GenuineIntel"
     "cuda:0"   : "NVIDIA GeForce RTX 3070" (8 GiB, sm_86, mempool enabled)
   Kernel cache:
     C:\Users\TAO\AppData\Local\NVIDIA\warp\Cache\1.6.2
Module __main__ e4b6b95 load on device 'cuda:0' took 25.43 ms  (cached)
Traceback (most recent call last):
  File "E:\clang\warp-162\warp\warp\examples\optim\example_fluid_checkpoint.py", line 442, in <module>
    print(f"Current memory usage: {wp.get_mempool_used_mem_current(device) / (1024 * 1024 * 1024):.4f} GiB")
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'warp' has no attribute 'get_mempool_used_mem_current'

### System Information

Windows 11
Warp 1.6.2
CUDA Toolkit 12.8, Driver 12.7

shi-eric, 2025-03-26:
This uses an API that will be released in Warp 1.7 next week. For now you can build from source or install a nightly build (see README.md)

closed 2025-03-26 by OmidGhotbi

shi-eric, 2025-03-26:
Great!