NVIDIA/warp #602: [BUG] Failed to compile tile_cholesky

arpit15, 2025-03-28:
### Bug Description

I am trying to run an example in mujoco_warp. I am getting error related to `tile_choleksy` function. I tried to build warp from scratch with libmathdx and still keep getting this error. 
Is their a way to debug this further? 

<details>
libmathdx cuSOLVER error: 3 on /home/user/projects/warp/warp/native/warp.cu:3272
Module mujoco_warp._src.solver._update_gradient 83bb4cc load on device 'cuda:0' took 343.87 ms
Traceback (most recent call last):
  File "/home/user/projects/warp/warp/codegen.py", line 1020, in build
    adj.eval(adj.tree.body[0])
  File "/home/user/projects/warp/warp/codegen.py", line 2777, in eval
    return emit_node(adj, node)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 1674, in emit_FunctionDef
    adj.eval(f)
  File "/home/user/projects/warp/warp/codegen.py", line 2777, in eval
    return emit_node(adj, node)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 2566, in emit_Assign
    rhs = adj.eval(node.value)
          ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 2777, in eval
    return emit_node(adj, node)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 2298, in emit_Call
    out = adj.add_call(func, args, kwargs, type_args, min_outputs=min_outputs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 1409, in add_call
    func_args, template_args, ltoirs, extra_shared_memory = func.lto_dispatch_func(
                                                            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/builtins.py", line 6605, in tile_cholesky_generic_lto_dispatch_func
    lto_symbol, lto_code_data = warp.build.build_lto_solver(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/build.py", line 423, in build_lto_solver
    raise RuntimeError("Failed to compile tile_cholesky")
RuntimeError: Failed to compile tile_cholesky

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/projects/mujoco_warp/.venv/bin/mjwarp-testspeed", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/testspeed.py", line 207, in main
    app.run(_main)
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.11/site-packages/absl/app.py", line 316, in run
    _run_main(main, args)
  File "/home/user/projects/mujoco_warp/.venv/lib/python3.11/site-packages/absl/app.py", line 261, in _run_main
    sys.exit(main(argv))
             ^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/testspeed.py", line 131, in _main
    jit_time, run_time, trace, ncon, nefc = mjwarp.benchmark(
                                            ^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/test_util.py", line 69, in benchmark
    fn(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/forward.py", line 684, in step
    forward(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/forward.py", line 678, in forward
    solver.solve(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/warp_util.py", line 101, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/solver.py", line 1002, in solve
    _create_context(m, d, grad=True)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/solver.py", line 69, in _create_context
    _update_gradient(m, d)
  File "/home/user/projects/mujoco_warp/mujoco_warp/_src/solver.py", line 296, in _update_gradient
    wp.launch(_zero_grad_dot, dim=(d.nworld), inputs=[d])
  File "/home/user/projects/warp/warp/context.py", line 5723, in launch
    module_exec = kernel.module.load(device, block_dim)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/context.py", line 2186, in load
    builder = ModuleBuilder(self, builder_options, hasher=self.hashers[active_block_dim])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/context.py", line 1692, in __init__
    self.build_kernel(kernel)
  File "/home/user/projects/warp/warp/context.py", line 1721, in build_kernel
    kernel.adj.build(self)
  File "/home/user/projects/warp/warp/codegen.py", line 1031, in build
    raise e
  File "/home/user/projects/warp/warp/codegen.py", line 1020, in build
    adj.eval(adj.tree.body[0])
  File "/home/user/projects/warp/warp/codegen.py", line 2777, in eval
    return emit_node(adj, node)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 1674, in emit_FunctionDef
    adj.eval(f)
  File "/home/user/projects/warp/warp/codegen.py", line 2777, in eval
    return emit_node(adj, node)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 2566, in emit_Assign
    rhs = adj.eval(node.value)
          ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 2777, in eval
    return emit_node(adj, node)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 2298, in emit_Call
    out = adj.add_call(func, args, kwargs, type_args, min_outputs=min_outputs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/codegen.py", line 1409, in add_call
    func_args, template_args, ltoirs, extra_shared_memory = func.lto_dispatch_func(
                                                            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/builtins.py", line 6605, in tile_cholesky_generic_lto_dispatch_func
    lto_symbol, lto_code_data = warp.build.build_lto_solver(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/build.py", line 423, in build_lto_solver
    raise RuntimeError("Failed to compile tile_cholesky")
RuntimeError: Error while parsing function "_cholesky" at /home/user/projects/mujoco_warp/mujoco_warp/_src/solver.py:290:
  fact_tile = wp.tile_cholesky(mat_tile)
;Failed to compile tile_cholesky

</details>

### System Information

OS: Ubuntu 22.04 
CUDA: 12.8
Nvidia Driver: 560.35.05
python: 3.11.9
g++ - 11.4.0
Warp - 1.7.0-rc.4 (`d7e9b9e641f7b2ad4c4056e9dfb30ad9f4c40c0a`)

shi-eric, 2025-03-28:
What GPU are you running with? We should be able to get more information if you run with `LIBMATHDX_LOG_LEVEL=5` set in your environment.

arpit15, 2025-03-28:
Thanks. Shared mem issue as shown below. Maybe will post in warp

<details>
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverCreate] handle=0x7ffc078ab980
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64Array] handle=0 op=CUSOLVER_OPERATOR_SIZE count=2 array=570,570
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64Array] handle=0 op=CUSOLVER_OPERATOR_BLOCK_DIM count=3 array=256,1,1
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_TYPE value=0
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_API value=0
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_FUNCTION value=2
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_EXECUTION value=1
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_PRECISION value=5
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_FILL_MODE value=0
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOperatorInt64] handle=0 op=CUSOLVER_OPERATOR_SM value=890
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverSetOptionStr] handle=0 opt=0 value=potrf_570_570_89_5
[2025-03-27 18:02:37][libmathdx][3010371][Api][cusolverGetLTOIRSize] handle=0 lto_size=0x7ffc078ab988
[2025-03-27 18:02:38][libmathdx][3010371][Info][cusolverGetLTOIRSize] Decompression done in 59.573 ms.
[2025-03-27 18:02:38][libmathdx][3010371][Info][cusolverGetLTOIRSize] NVRTC 12.8 compiling for SM 8.9
[2025-03-27 18:02:38][libmathdx][3010371][Info][cusolverGetLTOIRSize] NVRTC options --relocatable-device-code=true,--device-as-default-execution-space,--std=c++17,-DCUFFTDX_DETAIL_USE_CUDA_STL=1,-dlto,--gpu-architecture=compute_89
[2025-03-27 18:02:38][libmathdx][3010371][Info][cusolverGetLTOIRSize] NVRTC compilation finished in 250.019 ms.
[2025-03-27 18:02:38][libmathdx][3010371][Error][cusolverGetLTOIRSize] NVRTC compilation failed, error message:
cusolverdx/detail/solver_description.hpp(183): error: static assertion failed with "Provided combination of data type and sizes makes this problem not fit into shared memory available on the specified architecture"
              static_assert((not has_sm) or valid_shared_size,
              ^
          detected during instantiation of class "cusolverdx::detail::solver_description<Operators...> [with Operators=<cusolverdx::Size<570U, 570U, 1U>, cusolverdx::Precision<float, float, float>, cusolverdx::Type<(cusolverdx::type)0>, cusolverdx::Function<cusolverdx::function::potrf>, cusolverdx::SM<890U>>]" at line 3 of 68765bc8664acee4.cu

cusolverdx/detail/solver_description.hpp(183): error: static assertion failed with "Provided combination of data type and sizes makes this problem not fit into shared memory available on the specified architecture"
              static_assert((not has_sm) or valid_shared_size,
              ^
          detected during instantiation of class "cusolverdx::detail::solver_description<Operators...> [with Operators=<cusolverdx::Size<570U, 570U, 1U>, cusolverdx::Precision<float, float, float>, cusolverdx::Type<(cusolverdx::type)0>, cusolverdx::Function<cusolverdx::function::potrf>, cusolverdx::SM<890U>, cusolverdx::FillMode<cusolverdx::fill_mode::upper>>]" at line 3 of 68765bc8664acee4.cu

cusolverdx/detail/solver_description.hpp(183): error: static assertion failed with "Provided combination of data type and sizes makes this problem not fit into shared memory available on the specified architecture"
              static_assert((not has_sm) or valid_shared_size,
              ^
          detected during instantiation of class "cusolverdx::detail::solver_description<Operators...> [with Operators=<cusolverdx::Size<570U, 570U, 1U>, cusolverdx::Precision<float, float, float>, cusolverdx::Type<(cusolverdx::type)0>, cusolverdx::Function<cusolverdx::function::potrf>, cusolverdx::SM<890U>, cusolverdx::FillMode<cusolverdx::fill_mode::upper>, cusolverdx::BlockDim<256U, 1U, 1U>>]" at line 3 of 68765bc8664acee4.cu

cusolverdx/detail/solver_description.hpp(183): error: static assertion failed with "Provided combination of data type and sizes makes this problem not fit into shared memory available on the specified architecture"
              static_assert((not has_sm) or valid_shared_size,
              ^
          detected during:
            instantiation of class "cusolverdx::detail::solver_description<Operators...> [with Operators=<cusolverdx::Size<570U, 570U, 1U>, cusolverdx::Precision<float, float, float>, cusolverdx::Type<(cusolverdx::type)0>, cusolverdx::Function<cusolverdx::function::potrf>, cusolverdx::SM<890U>, cusolverdx::FillMode<cusolverdx::fill_mode::upper>, cusolverdx::BlockDim<256U, 1U, 1U>, cusolverdx::Block>]" at line 29 of cusolverdx/detail/solver_execution.hpp
            instantiation of class "cusolverdx::detail::solver_execution<Operators...> [with Operators=<cusolverdx::Size<570U, 570U, 1U>, cusolverdx::Precision<float, float, float>, cusolverdx::Type<(cusolverdx::type)0>, cusolverdx::Function<cusolverdx::function::potrf>, cusolverdx::SM<890U>, cusolverdx::FillMode<cusolverdx::fill_mode::upper>, cusolverdx::BlockDim<256U, 1U, 1U>, cusolverdx::Block>]" at line 67 of cusolverdx/detail/solver_execution.hpp
[... cut ...]
  File "/home/user/projects/warp/warp/builtins.py", line 6605, in tile_cholesky_generic_lto_dispatch_func
    lto_symbol, lto_code_data = warp.build.build_lto_solver(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/projects/warp/warp/build.py", line 423, in build_lto_solver
    raise RuntimeError("Failed to compile tile_cholesky")
RuntimeError: Error while parsing function "_cholesky" at /home/user/projects/mujoco_warp/mujoco_warp/_src/solver.py:290:
  fact_tile = wp.tile_cholesky(mat_tile)
;Failed to compile tile_cholesky

</details>

arpit15, 2025-03-28:
I am trying a scene which has 98 boxes. My VRAM 24GB. I was expecting this to work on my GPU. Can it be a GPU issue or a parameter which I can set in warp?

shi-eric, 2025-03-28:
Thanks for the information, will ask around next week, but please raise this issue in https://github.com/google-deepmind/mujoco_warp so the team can track it there. Were you running an example straight from the repo, or did you modify it? It might help to have the command you're running.

arpit15, 2025-03-28:
I am running `mjwarp-testspeed` with my test scene which only has a bunch of boxes.

shi-eric, 2025-03-28:
Got it. Shared memory is a limited resource, so there will be a limit on the tile sizes that can be used depending on what's going on inside the kernel. The way it's being determined in mujoco_warp might be pretty naive right now and without bound (can't tell exactly how much from a quick look at your logs).

arpit15, 2025-03-28:
I reduced the number of boxes to 18 and things seem to work fine. I guess I will read the code to see if I can fix something to make it work.

arpit15, 2025-03-28:
I have posted the issue in (mujoco_warp repo)[https://github.com/google-deepmind/mujoco_warp/issues/47]. Please let me know if there is a quick hack to this issue.

shi-eric, 2025-03-31:
I think we determined that this isn't a bug with the Warp library, so I'm closing the issue. I think we can improve the error reporting, so I opened #608 to track this work.

closed 2025-03-31 by shi-eric