NVIDIA/warp #603: [REQ] Change string repr for warp.array

arpit15, 2025-03-28:
### Description
Currently generating kernel with `warp.array` requires using its `__str__` func which calls `.numpy()` over the array leading to data transfer. I propose to create this info from the class members like shape and dtype instead. 

I am happy to contribute. But I don't understand the internals of what is required for the JIT step from the `warp.array` 

### Context
This prevents from running graph capture with array involved. I encountered [this](https://github.com/google-deepmind/mujoco_warp/issues/17#issuecomment-2759715574) issue while running mujoco_warp example.

arpit15, 2025-03-28:
Oh weird. Is there a better way to debug? I just get a segfault and no other information.

shi-eric, 2025-03-28:
In the past, issues with graph capture are usually due to module compilation during graph capture. I don't think they are usually associated with a segfault, though. The observation you made about the MJWarp example not segfaulting without graph capture is also new to me.

If you want to rule out the module compilation issue, you could:

1. Run the example without graph capture.
2. Run the example with graph capture, but make sure not to clear the kernel cache

This should prevent any module compilation in step 2.

You can also set `faulthandler.enable()` to get some stack traces: https://docs.python.org/3/library/faulthandler.html

Then you can also try running with `cuda-gdb` (maybe with `wp.config.mode = "debug") to get more information.

arpit15, 2025-03-28:
I tried to rule out module compilation issue. Got the same error with all kernels `(cached)` in the second run

I added a `faulthandler.enable()` and `wp.config.mode = "debug"` . Still only see 
```
Extension modules: numpy._core._multiarray_umath, numpy.linalg._umath_linalg (total: 2)
Segmentation fault (core dumped)
``` 
I will need to check how to run python program with `cuda-gdb`

shi-eric, 2025-03-29:
> I tried to rule out module compilation issue. Got the same error with all kernels `(cached)` in the second run
> 
> I added a `faulthandler.enable()` and `wp.config.mode = "debug"` . Still only see
> 
> ```
> Extension modules: numpy._core._multiarray_umath, numpy.linalg._umath_linalg (total: 2)
> Segmentation fault (core dumped)
> ```
> 
> I will need to check how to run python program with `cuda-gdb` **Update** I tried to follow [this](https://stackoverflow.com/a/75488797) for running with `cuda-gdb`. I tried to run without breakpoints. I didn't see any logs. [@shi-eric](https://github.com/shi-eric) could you kindly post instructions on how to run with `cuda-gdb` ?

You can just do something like `cuda-gdb --args <your_original_run_command>` then type `run` and when it crashes you can type `bt` or `bt full` to try to get a stack trace.

arpit15, 2025-03-31:
I get following `bt` and I also had `wp.config.mode = "debug"` and `wp.config.debug = True` on. Do you see any information which is actionable in this?
```
#0  0x00007fff7de00279 in ?? () from /lib/x86_64-linux-gnu/libcuda.so
#1  0x00007fff7e09a2c0 in ?? () from /lib/x86_64-linux-gnu/libcuda.so
#2  0x00007fff7dd04e57 in ?? () from /lib/x86_64-linux-gnu/libcuda.so
#3  0x00007fff7dd0a1c7 in ?? () from /lib/x86_64-linux-gnu/libcuda.so
#4  0x00007fff7dedfbf0 in ?? () from /lib/x86_64-linux-gnu/libcuda.so
#5  0x00007fff846f4b2f in cuda_launch_kernel () from /home/ad.dex.ai/arpit.agarwal/projects/mujoco_warp/.venv/lib/python3.11/site-packages/warp/bin/warp.so
#6  0x00007ffff6405dae in ffi_call_unix64 () from /opt/dex/python/3.11.9/lib/python3.11/lib-dynload/../../libffi.so.6
#7  0x00007ffff640571f in ffi_call () from /opt/dex/python/3.11.9/lib/python3.11/lib-dynload/../../libffi.so.6
#8  0x00007ffff6a13645 in _call_function_pointer (argtypecount=<optimized out>, argcount=8, resmem=0x7fffffffc7c0, restype=<optimized out>, atypes=<optimized out>, avalues=<optimized out>, pProc=0x7fff846f4a30 <cuda_launch_kernel>, flags=4353)
    at /home/dexterity/.cache/pyenv/build/3.11.9/Python-3.11.9/Modules/_ctypes/callproc.c:923
#9  _ctypes_callproc (pProc=0x7fff846f4a30 <cuda_launch_kernel>, argtuple=0x7ffff37099a0, flags=4353, argtypes=<optimized out>, restype=<optimized out>, checker=0x0) at /home/dexterity/.cache/pyenv/build/3.11.9/Python-3.11.9/Modules/_ctypes/callproc.c:1262
#10 0x00007ffff6a13c8f in PyCFuncPtr_call (self=0x7ffff38bedd0, inargs=0x7ffff37099a0, kwds=<optimized out>) at /home/dexterity/.cache/pyenv/build/3.11.9/Python-3.11.9/Modules/_ctypes/_ctypes.c:4201
#11 0x000055555560509d in _PyObject_MakeTpCall (tstate=0x555555b8eed8 <_PyRuntime+166328>, callable=0x7ffff38bedd0, args=<optimized out>, nargs=<optimized out>, keywords=0x0) at Objects/call.c:214
#12 0x00005555555ee007 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=<optimized out>, args=<optimized out>, callable=0x7ffff38bedd0, tstate=<optimized out>) at ./Include/internal/pycore_call.h:90
#13 PyObject_Vectorcall (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=0x7ffff38bedd0) at Objects/call.c:299
#14 _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:4769
#15 0x000055555550f76d in _PyEval_EvalFrame (throwflag=0, frame=0x7ffff7fb07e0, tstate=0x555555b8eed8 <_PyRuntime+166328>) at ./Include/internal/pycore_ceval.h:73
#16 _PyEval_Vector (tstate=0x555555b8eed8 <_PyRuntime+166328>, func=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>) at Python/ceval.c:6434
#17 0x00005555555f96c8 in _PyVectorcall_Call (kwargs=<optimized out>, tuple=0x7ffff2befc00, callable=0x7ffff457fce0, func=0x5555555f9620 <_PyFunction_Vectorcall>, tstate=0x555555b8eed8 <_PyRuntime+166328>) at Objects/call.c:245
#18 _PyObject_Call (tstate=0x555555b8eed8 <_PyRuntime+166328>, callable=0x7ffff457fce0, args=0x7ffff2befc00, kwargs=<optimized out>) at Objects/call.c:328
#19 0x00005555555f1e9c in PyObject_Call (kwargs=0x7ffff2c221c0, args=0x7ffff2befc00, callable=0x7ffff457fce0) at Objects/call.c:355
#20 do_call_core (use_tracing=<optimized out>, kwdict=0x7ffff2c221c0, callargs=0x7ffff2befc00, func=0x7ffff457fce0, tstate=<optimized out>) at Python/ceval.c:7349
#21 _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:5376
#22 0x000055555550f76d in _PyEval_EvalFrame (throwflag=0, frame=0x7ffff7fb0698, tstate=0x555555b8eed8 <_PyRuntime+166328>) at ./Include/internal/pycore_ceval.h:73
#23 _PyEval_Vector (tstate=0x555555b8eed8 <_PyRuntime+166328>, func=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>) at Python/ceval.c:6434
#24 0x00005555555f96c8 in _PyVectorcall_Call (kwargs=<optimized out>, tuple=0x7ffff38e94c0, callable=0x7ffff3e039c0, func=0x5555555f9620 <_PyFunction_Vectorcall>, tstate=0x555555b8eed8 <_PyRuntime+166328>) at Objects/call.c:245
#25 _PyObject_Call (tstate=0x555555b8eed8 <_PyRuntime+166328>, callable=0x7ffff3e039c0, args=0x7ffff38e94c0, kwargs=<optimized out>) at Objects/call.c:328
#26 0x00005555555f1e9c in PyObject_Call (kwargs=0x7ffff2bc7f40, args=0x7ffff38e94c0, callable=0x7ffff3e039c0) at Objects/call.c:355
#27 do_call_core (use_tracing=<optimized out>, kwdict=0x7ffff2bc7f40, callargs=0x7ffff38e94c0, func=0x7ffff3e039c0, tstate=<optimized out>) at Python/ceval.c:7349
#28 _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:5376
#29 0x000055555550f76d in _PyEval_EvalFrame (throwflag=0, frame=0x7ffff7fb0550, tstate=0x555555b8eed8 <_PyRuntime+166328>) at ./Include/internal/pycore_ceval.h:73
#30 _PyEval_Vector (tstate=0x555555b8eed8 <_PyRuntime+166328>, func=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>) at Python/ceval.c:6434
#31 0x00005555555f96c8 in _PyVectorcall_Call (kwargs=<optimized out>, tuple=0x7ffff38ff400, callable=0x7ffff3e03ec0, func=0x5555555f9620 <_PyFunction_Vectorcall>, tstate=0x555555b8eed8 <_PyRuntime+166328>) at Objects/call.c:245
#32 _PyObject_Call (tstate=0x555555b8eed8 <_PyRuntime+166328>, callable=0x7ffff3e03ec0, args=0x7ffff38ff400, kwargs=<optimized out>) at Objects/call.c:328
[... cut ...]
#52 pymain_run_file (config=0x555555b74f20 <_PyRuntime+59904>) at Modules/main.c:379
#53 pymain_run_python (exitcode=0x7fffffffd5c0) at Modules/main.c:601
#54 Py_RunMain () at Modules/main.c:680
#55 0x0000555555748cd9 in Py_BytesMain (argc=<optimized out>, argv=<optimized out>) at Modules/main.c:734
#56 0x00007ffff7c29d90 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#57 0x00007ffff7c29e40 in __libc_start_main () from /lib/x86_64-linux-gnu/libc.so.6
--Type <RET> for more, q to quit, c to continue without paging--
#58 0x00005555556dcf3a in _start ()
```

shi-eric, 2025-03-31:
Thanks for running this. I don't have anything to suggest that would be a good use of time, but it seems similar to the segfault issues encountered in #372 and #552. It seems like we might not be doing something right when it comes to a calling convention mismatch with `ctypes`. I plan to spend some time working on the issue soon.

Do you know how you got your Python installation? You can also try to set up an environment with Conda's Python installations or `uv`'s. You might get different results.

`wp.config.debug = True` doesn't do anything.

arpit15, 2025-04-09:
I used poetry for my python installation. I am using python 3.11. Can you kindly provide your python env configs like python version,  conda version, etc. for testing.
I am wondering if you got chance to work on this

shi-eric, 2025-04-09:
Here's an environment file for an environment I just tested successfully with (I'm using https://github.com/conda-forge/miniforge here): 

```
name: mujoco_warp_conda_py11
channels:
  - conda-forge
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=2_gnu
  - bzip2=1.0.8=h4bc722e_7
  - ca-certificates=2025.1.31=hbcca054_0
  - ld_impl_linux-64=2.43=h712a8e2_4
  - libexpat=2.7.0=h5888daf_0
  - libffi=3.4.6=h2dba641_1
  - libgcc=14.2.0=h767d61c_2
  - libgcc-ng=14.2.0=h69a702a_2
  - libgomp=14.2.0=h767d61c_2
  - liblzma=5.8.1=hb9d3cd8_0
  - libnsl=2.0.1=hd590300_0
  - libsqlite=3.49.1=hee588c1_2
  - libuuid=2.38.1=h0b41bf4_0
  - libxcrypt=4.4.36=hd590300_1
  - libzlib=1.3.1=hb9d3cd8_2
  - ncurses=6.5=h2d0b736_3
  - openssl=3.5.0=h7b32b05_0
  - pip=25.0.1=pyh8b19718_0
  - python=3.11.11=h9e4cc4f_2_cpython
  - readline=8.2=h8c095d6_2
  - setuptools=78.1.0=pyhff2d567_0
  - tk=8.6.13=noxft_h4845f30_101
  - tzdata=2025b=h78e105d_0
  - wheel=0.45.1=pyhd8ed1ab_1
  - pip:
      - absl-py==2.2.2
      - etils==1.12.2
      - fsspec==2025.3.2
      - glfw==2.8.0
      - importlib-resources==6.5.2
      - iniconfig==2.1.0
      - mujoco==3.3.0
      - mujoco-warp==3.2.7
      - numpy==2.2.4
      - packaging==24.2
      - pluggy==1.5.0
      - pyopengl==3.1.9
      - pytest==8.3.5
      - typing-extensions==4.13.1
      - warp-lang==1.8.0.dev20250409
      - zipp==3.21.0
prefix: /home/eshi/miniforge3/envs/mujoco_warp_conda_py11
```

closed 2025-04-21 by eric-heiden