NVIDIA/warp #604: [QUESTION] Calling fem.integrate inside LinearOperator.matvec for matrix-free CG?

akrivx, 2025-03-28:
Hi all,

I'm experimenting with a matrix-free CG solver and wanted to ask whether this is a reasonable approach or a misuse of `warp.fem`.

I'm currently defining a `LinearOperator` where the `matvec` performs a `wp.fem.integrate(...)` call to evaluate the action of a bilinear form at each iteration. Here's a simplified example:

```python
@fem.integrand
def newton_lhs_form(s: fem.Sample, u_cur: fem.Field, v_cur: fem.Field, u: fem.Field, v: fem.Field, dt: float, density: float, alpha: float, beta: float, lame: wp.vec2, scl: float):
    # ...

def make_matrix_free_newton_linear_operator(self):
    x_field = self.u_space.make_field()

    def matvec(x: wp.array, y: wp.array, z: wp.array, alpha: float, beta: float):
        """Compute z = alpha * A @ x + beta * y"""
        x_field.dof_values = x
        z.zero_()
        fem.integrate(
            newton_lhs_form,
            quadrature=self.u_quadrature,
            fields={
                'u_cur': self.u_field,
                'v_cur': self.v_field,
                'u': x_field,
                'v': self.u_test,
            },
            values={
                'dt': self.dt,
                'density': self.density,
                'alpha': self.damping_alpha,
                'beta': self.damping_beta,
                'lame': self.lame,
                'scl': alpha,
            },
            output=z,
            add=True
        )
        linalg.array_axpy(x=y, y=z, alpha=beta, beta=1.0)

    return LinearOperator(
        shape=self.u_bd_matrix.shape,
        dtype=self.u_bd_matrix.dtype,
        device=self.u_bd_matrix.device,
        matvec=matvec
    )
```

It works, but it's significantly slower than directly assembling and using the bilinear form (I can quantify that if needed), so I'm concerned:
- Is it reasonable to call `integrate(...)` this way in every CG iteration?
- Am I creating unnecessary overhead by introducing `x_field.dof_values` each time, and am I actually misusing fields here?
- Are there better matrix-free strategies in `warp.fem` I might be missing?

I’d love to understand if this pattern is encouraged or if I should lean toward assembling the matrix and preconditioning instead.

Thanks,
Chris

akrivx, 2025-03-31:
I noticed that `cg` can optionally capture the solver iteration as a CUDA graph to reduce launch overhead with the `use_cuda_graph` parameter (which I set to `True`; according to the docstring, the linear operator and preconditioner must only perform graph-friendly operations, which I'm unsure how to interpret).

I'll profile things with `nsys` as suggested. Meanwhile, for what it's worth, here's a brief performance analysis:

## Overview

I benchmarked the performance of matrix-free (MF) versus assembled conjugate gradient (CG) solves in `warp.fem` using a 2.5D hexahedral mesh (i.e., $$RxRx1$$, where $$R$$ is the mesh resolution) with a linear function space, 2nd-order quadrature, and 3 DOFs per node. Each solve used a fixed number of 200 CG iterations to ensure that timing differences are purely due to operator evaluation cost.

## System

```
Warp 1.6.2 initialized:
   CUDA Toolkit 12.8, Driver 12.8
   Devices:
     "cpu"      : "AMD64 Family 26 Model 68 Stepping 0, AuthenticAMD"
     "cuda:0"   : "NVIDIA GeForce RTX 5080" (16 GiB, sm_120, mempool enabled)
```

## Results

| Resolution | Matrix-Free CG (ms) | Assembled CG (ms) | Slowdown Factor (MF / Assembled) |
| -- | -- | -- | -- |
| 10 | 9.88 | 5.79 | 1.71 |
| 20 | 11.51 | 5.86 | 1.96 |
| 40 | 22.64 | 6.82 | 3.32 |
| 80 | 61.96 | 8.03 | 7.72 |
| 160 | 220.21 | 14.19 | 15.52 |
| 320 | 873.27 | 85.37 | 10.23 |
| 640 | 3494.59 | 343.71 | 10.17 |

## Graphs

### CG Solve Time vs Mesh Resolution

[image]

- Matrix-free time increases rapidly with resolution
- Assembled CG time grows much more slowly

### Matrix-Free vs Assembled Slowdown Factor

[image]

- Slowdown grows quickly, peaking around 15×, then stabilising near 10×

## Key Findings

- At low resolutions, matrix-free CG is only ~2× slower
- But at higher resolutions, the cost grows significantly:
  - 10× slower at 320 and 640
- Up to 160, matrix-free CG grows more rapidly with mesh resolution than assembled CG
- Between 160 and 320, assembled CG experiences a sharper increase in solve time
- Beyond 320, both approaches scale similarly with respect to mesh resolution

All CG iterations were fixed — differences are purely per-matvec cost

gdaviet, 2025-03-31:
Thanks for running those tests! These non-linearities in the scaling are indeed surprising, I'll need to investigate more

gdaviet, 2025-03-31:
In general since the linear operator is going to be graph-captured it would be safer to explicitly copy the vector into the field: 
`wp.copy(src=x, dest=x_field.dof_values)` (the Python-side assignment won't be replayed when launching the graph). 
But in your case I don't think this matters, as the memory address of `x` is not going to change over iterations, so `fem.integrate` should still read the correct data.
Outside of graph-capturing concerns, I would say updating the `dof_values` is indeed the idiomatic way of updating the data passed to the integrand.

gdaviet, 2025-04-01:
I ran a few more tests, and mostly observe the same behavior as you reported. Graph capture seems to be working properly, the compute time is in the kernel, not the Python overhead.
What I can suggest to reduce the slowdown is to precompute and cache some of the quantities that do not depend on the CG iteration, for instance the SVD factorization. You can to that by use fem.interpolate and storing the result in arbitrary arrays using the sample's `qp_index`, for instance:

```
    @fem.integrand 
    def polar_decomposition(s: fem.Sample, u:fem.Field, Rs:wp.array(dtype=wp.mat33), Ss:wp.array(dtype=wp.mat33)):
        F = defgrad(u, s)        

        U = wp.mat33()
        D = wp.vec3()
        V = wp.mat33()
        wp.svd3(F, U, D, V)

        R = U @ wp.transpose(V)
        S = V @ wp.diag(D) @ wp.transpose(V)

        Rs[s.qp_index] = R
        Ss[s.qp_index] = S

    def make_matrix_free_newton_linear_operator(self):

        R = wp.empty(dtype=wp.mat33, shape=self.elasticity_quadrature.total_point_count())
        S = wp.empty(dtype=wp.mat33, shape=self.elasticity_quadrature.total_point_count())

        fem.interpolate(ClassicFEM.polar_decomposition, 
                      quadrature= self.elasticity_quadrature,
                      fields={"u": self.u_field},
                      values = {"Rs": R, "Ss": S}
                      )

        x_field = self.u_field.space.make_field()
...
```

then read back from those arrays in the bilinear form

akrivx, 2025-04-01:
Yes, that sounds like an excellent idea; thank you. I'll give it a try and let you know soon.