NVIDIA/warp #618: [QUESTION] How to apply displacement field to update mesh geometry in `warp.fem`?

akrivx, 2025-04-03:
I've computed a displacement field for my simulation using `warp.fem`, but I'm unclear about the correct way to apply this displacement to update my undeformed mesh geometry.

Initially, I tried using `fem.interpolate` with `PicQuadrature` to map displacement fields back onto mesh vertices. However, I realised this interpolation approach only works for structured grids (`fem.Grid2D`, `fem.Grid3D`), not for general meshes.

Given that there isn't necessarily a one-to-one correspondence between mesh vertices and polynomial space nodes, what's the recommended way to interpolate or otherwise apply displacement fields computed in arbitrary polynomial spaces back onto a general (possibly unstructured) mesh?

Any guidance or example workflows would be greatly appreciated.

Thanks,
Chris

akrivx, 2025-04-04:
I think I figured out a possible solution:

```python
@fem.integrand
def deformed_position_integrand(s: fem.Sample, domain: fem.Domain, displacement: fem.Field, rest_pos: wp.array(dtype=wp.vec3)):
    i = s.qp_index
    s = fem.lookup(domain, rest_pos[i])
    return rest_pos[i] + displacement(s)

deformed_pos = wp.empty_like(pos)
fem.interpolate(
    deformed_position_integrand,
    dest=deformed_pos,
    domain=domain,
    dim=pos.shape[0],
    fields={'displacement': displacement},
    values={'rest_pos': pos},
)
```

For the above to work, the underlying mesh must be constructed with `build_bvh=True`.

However, this solution is not compatible with `Hexmesh` because `Hexmesh` does not support the `fem.lookup` operator. 

Interestingly, passing `build_bvh=True` to `Trimesh3D` won't work either:
```bash
RuntimeError: Error launching kernel 'Trimesh___compute_tri_bounds', argument 'positions' expects an array with dtype=<class 'warp.types.vec2f'> but passed array has dtype=<class 'warp.types.vec3f'>.
```

Is this a known limitation or a bug?

gdaviet, 2025-04-07:
Sorry for the late answer. Yes, using `PicQuadrature` (which uses `lookup` internally) is the preferred way to do interpolation at arbitrary points. `Hexmesh`/`Quadmesh` no supporting `lookup` is a known limitation that I've been  planning to address for a while, as well as being able to customize the behavior when the looked-up point does not belong to any element.

In the meantime, it is possible to define a `PicQuadrature` while explicit cell indices and coordinates instead of global positions, using you own custom `lookup` kernel. For instance, using a Newton loop to find the local coordinates corresponding to a global position:
```   v_pos = vertex_pos[v]

    # dX/dc
    coords = s.element_coords
    for k in range(64):
        s = fem.types.make_free_sample(s.element_index, coords)
        pos = domain(s)
        F = fem.deformation_gradient(domain, s)

        coords += 0.25 * wp.inverse(F) * (v_pos - pos)
        coords = wp.vec3(
            wp.clamp(coords[0], 0.0, 1.0),
            wp.clamp(coords[1], 0.0, 1.0),
            wp.clamp(coords[2], 0.0, 1.0),
        )
```

akrivx, 2025-04-07:
No worries, thanks for the response.

What about triangle meshes? Shouldn't the approach in my last comment work? Is the runtime error I'm seeing expected?

akrivx, 2025-04-07:
Excellent, thank you.

shi-eric, 2025-04-10:
Fixed by @gdaviet in 6f9411f34002e5b8be3fce32987957c252de9120