NVIDIA/warp #621: [BUG] The add_cloth_mesh adds a non-existing spring which makes the XPBD integrator explode

ManosAgelidis, 2025-04-04:
### Bug Description

When loading an external mesh as a cloth mesh and passing the add_springs = True parameters to the add_cloth_mesh function a non-existing spring between vertices and 0,0,0 seems to be added. To reproduce you can try the following example where I have modified the example_cloth.py example to use the XPBD integrator and to load a cloth mesh rather than create a cloth grid. The behaviour can be seen in the video, sorry for the low resolution, github didn't allow me to load the full video.

[image]

The problem seems to be coming from these lines in the add_cloth_mesh function in model.py

if add_springs:
            spring_indices = set()
            for i, j, k, l in edge_indices:
                spring_indices.add((min(i, j), max(i, j)))
                spring_indices.add((min(i, k), max(i, k)))
                spring_indices.add((min(i, l), max(i, l)))

                spring_indices.add((min(j, k), max(j, k)))
                spring_indices.add((min(j, l), max(j, l)))

                spring_indices.add((min(k, l), max(k, l)))

            for i, j in spring_indices:
                self.add_spring(i, j, spring_ke, spring_kd, control=0.0)

 

### System Information

Warp version : '1.7.0'
Python version: 3.12.8 
OS: Windows 11

AnkaChan, 2025-04-04:
**Hi Manos,

Could you please provide a minimal code example that reproduces this issue? That would greatly help me investigate it further.

AnkaChan, 2025-04-06:
Nevermind I think I have figured out the issue

ManosAgelidis, 2025-04-07:
Hi Anka, thanks for the update. Let me know if you need a full script.

shi-eric, 2025-04-16:
Thanks @ManosAgelidis, @AnkaChan fixed this in 37079d1fe5101ebf205f8f87f8ab6160e391b6d8. This fix will be in the Warp 1.7.1 release at the beginning of May, or you can try it out with the nightly builds / build from source. Please verify that it fixes the problem you reported!

closed 2025-04-16 by AnkaChan

ManosAgelidis, 2025-04-24:
Thanks @AnkaChan , @shi-eric, tested and it works perfectly!