NVIDIA/warp #624: [DOCS] wp.array auto-conversion using wp.array()

JonathanKuelz, 2025-04-04:
## Category

- [ ] Report an error in the documentation.
- [x] Request for something to be documented.
- [ ] Suggestion to improve the documentation.
- [x] Other: Warning / error message

## Description
The documentation recommends constructing warp arrays from torch (I guess the same holds for numpy) using `from_torch`. However, it is also *possible* to do the following:

```
a = torch.arange(1000)
wp1 = wp.from_torch(t, dtype=wp.int32)
wp2 = wp.from_torch(t, dtype=wp.int64)
```

While this does not seem to be recommended anywhere, I suggest raising a warning or even an error if this is done and it triggers a data type conversion.

The above example will result in:
```
wp1 -> [0 0 1 0 2 0 3 0 4 0]
wp2 -> [0 1 2 3 4 5 6 7 8 9]
```
without any warning, which can lead to hard-to-detect bugs.

shi-eric, 2025-04-08:
Would it help if we put the warning at a higher verbosity level, e.g. if you set `wp.config.verbose = True`? I don't want to presume that everyone doing a non-standard type conversion would like to be warned by default. I don't know if setting `wp.config.verbose = True` is part of your debugging workflow, though.

JonathanKuelz, 2025-04-08:
Sure - it would have already helped me if it was mentioned in the Docs, e.g. [here](https://nvidia.github.io/warp/modules/interoperability.html). Printing only if `wp.config.verbose = True` seems like a good solution, too.

shi-eric, 2025-05-04:
Can you please clarify the code that produced the output in the original comment?

Input:

```python
import torch

import warp as wp

wp.init()

a = torch.arange(10)
wp1 = wp.from_torch(a, dtype=wp.int32)
wp2 = wp.from_torch(a, dtype=wp.int64)

print(wp1)
print(wp2)
```

```
RuntimeError: Cannot convert Torch type torch.int64 to Warp type <class 'warp.types.int32'>
```

JonathanKuelz, 2025-05-04:
I'm sorry, apparently I made a mistake when copying my own code. The original issue is not clear in what I wanted to communicate. Here is an example with code that actually reproduces the issue I wanted talked about: Using `from_torch` works as intended. Using `wp.array` directly does not always give the expected results but also doesn't fail (as `from_torch` does).

```python
import torch
import warp as wp

wp.init()
t_cpu = torch.arange(10, device='cpu')
t_gpu = torch.arange(10, device='cuda')

wp_1_cpu = wp.array(t_cpu, dtype=wp.int32)
wp_1_gpu = wp.array(t_gpu, dtype=wp.int32)
wp_2_cpu = wp.array(t_cpu, dtype=wp.int64)
wp_2_gpu = wp.array(t_gpu, dtype=wp.int64)

# int32 will throw an exception
wp_2_cpu_from_torch = wp.from_torch(t_cpu, dtype=wp.int64)
wp_2_gpu_from_torch = wp.from_torch(t_gpu, dtype=wp.int64)

print("Conversion as intended")
print('int64, from torch, cpu:', wp_2_cpu_from_torch)
print('int64, from torch, gpu:', wp_2_gpu_from_torch)

print("Direct conversion to array:")
print('int32, cpu:', wp_1_cpu)
print('int32, gpu:', wp_1_gpu)
print('int64, cpu:', wp_2_cpu)
print('int64, gpu:', wp_2_gpu)
```

Which results in
```
Conversion as intended
int64, from torch, cpu: [0 1 2 3 4 5 6 7 8 9]
int64, from torch, gpu: [0 1 2 3 4 5 6 7 8 9]
Direct conversion to array:
int32, cpu: [0 1 2 3 4 5 6 7 8 9]
int32, gpu: [0 0 1 0 2 0 3 0 4 0]
int64, cpu: [0 1 2 3 4 5 6 7 8 9]
int64, gpu: [0 1 2 3 4 5 6 7 8 9]
```

shi-eric, 2025-05-04:
Thanks! I'll take a look.

shi-eric, 2025-05-04:
Here's what I have in a new branch (no need to flip on a verbose flag, this is the default behavior):

```
Warp UserWarning: The input data type int64 does not appear to be compatible with the requested dtype <class 'warp.types.int32'>
Conversion as intended
int64, from torch, cpu: [0 1 2 3 4 5 6 7 8 9]
int64, from torch, gpu: [0 1 2 3 4 5 6 7 8 9]
Direct conversion to array:
int32, cpu: [0 1 2 3 4 5 6 7 8 9]
int32, gpu: [0 0 1 0 2 0 3 0 4 0]
int64, cpu: [0 1 2 3 4 5 6 7 8 9]
int64, gpu: [0 1 2 3 4 5 6 7 8 9]
```

Let me know if this seems good to you.

JonathanKuelz, 2025-05-04:
Yes, I was looking for something like that when I opened the issues :)

shi-eric, 2025-05-16:
This change went in with 5e32b5f0b419834761860a4ef3a7600dbea97bb0

Eventually, we might raise an exception for certain conversions, e.g. requesting a 32-bit type when the Torch array is a 64-bit type.

closed 2025-05-31 by shi-eric