NVIDIA/warp #628: [QUESTION] How to properly use Warp-generated CUDA files in C++? Need more complete guidance

ChenlizheMe, 2025-04-07:
I'm trying to use NVIDIA Warp to generate CUDA code and then integrate it into my C++ project. I noticed in issue #102 that `warp.h` is mentioned as the interface, but as a beginner, I'm having trouble with the implementation.

ChenlizheMe, 2025-04-07:
> Hi, maybe this helps https://github.com/erwincoumans/warp_cpp . They first run a kernel that is compiled with warp and is stored in warp's cache, and then link against it via CMake

Apologies for the confusion earlier! After compiling Warp from source and reinstalling, I can now confirm that PTX files are correctly generated. It seems the issue was specific to the pip-installed version in my environment. Thank you for pointing me in the right direction—your help was crucial in resolving this!

shi-eric, 2025-04-07:
> > Hi, maybe this helps https://github.com/erwincoumans/warp_cpp . They first run a kernel that is compiled with warp and is stored in warp's cache, and then link against it via CMake
> 
> Apologies for the confusion earlier—I figured out the issue! After compiling Warp from source and reinstalling, the PTX files are now correctly generated. It seems the problem was related to the pre-built `pip` installation. Much appreciated!

There's an explicit configuration setting to control the output format: https://nvidia.github.io/warp/configuration.html#warp.config.cuda_output

ChenlizheMe, 2025-04-08:
> There's an explicit configuration setting to control the output format: https://nvidia.github.io/warp/configuration.html#warp.config.cuda_output

Thx！