NVIDIA/warp #629: [QUESTION] How to access vertices of both triangles sharing an edge in a `fem.integrand` function?

akrivx, 2025-04-07:
Hi,

I'm working with a triangle mesh and using `fem.Sides` to represent edges. I'd like to compute a scalar quantity per interior edge (i.e., an edge shared by two triangles) using `fem.interpolate`, and for that, I need access to the four vertices involved (ignoring boundary edges where the quantity is zero):

- The two vertices that form the shared edge, and
- The vertex opposite the edge in each of the two adjacent triangles.

Ideally, I'd like to retrieve these vertex indices in a consistent and orderly way (e.g., counter-clockwise).

I know that given a sample point on a side, I can map it to the adjacent triangles using `fem.to_inner_cell` / `fem.to_outer_cell`, but I can't figure out how to retrieve the vertex indices or coordinates for those elements or the edge itself.

Are there any examples showing how to do this, or suggestions for recovering this 4-vertex configuration from within a `fem.integrand`?

Thanks,
Chris

gdaviet, 2025-04-08:
Unfortunately there's no public, generic way to access this information for now, you will need to access the geometry internals (i.e, assume knowledge of how the "domain" argument is laid out).

Assuming that you are using the `Trimesh` class, what you can do is:

```
import warp as wp
import warp.examples.fem.utils as fem_example_utils
import warp.fem as fem

@wp.func
def find_other_vertex(
    tri_vertex_indices: wp.array2d(dtype=int),
    tri_idx: int,
    edge_vidx: wp.vec2i,
):
    tri_vidx = tri_vertex_indices[tri_idx]

    for k in range(3):
        v = tri_vidx[k]
        if v != edge_vidx[0] and v != edge_vidx[1]:
            return v

    return -1  # unreachable

@fem.integrand
def form(dmn: fem.Domain, s: fem.Sample):
    inner_s = fem.to_inner_cell(dmn, s)
    outer_s = fem.to_outer_cell(dmn, s)

    if inner_s.element_index == outer_s.element_index:
        # boundary side, ignore
        return

    # edge vertex indices
    # oriented such that (CCW) normal points towards outer cell
    edge_vidx = dmn.topology.edge_vertex_indices[s.element_index]

    cells = fem.cells(dmn)
    inner_v = find_other_vertex(cells.topology.tri_vertex_indices, inner_s.element_index, edge_vidx)
    outer_v = find_other_vertex(cells.topology.tri_vertex_indices, outer_s.element_index, edge_vidx)

    wp.printf("Edge %d: vertices %d %d %d %d \n", s.element_index, edge_vidx[0], inner_v, edge_vidx[1], outer_v)

resolution = 2
positions, tri_vidx = fem_example_utils.gen_trimesh(res=wp.vec2i(resolution))
geo = fem.Trimesh2D(tri_vertex_indices=tri_vidx, positions=positions)

sides = fem.Sides(geo)
fem.interpolate(form, quadrature=fem.RegularQuadrature(sides, order=0))
```

The non-portable parts are the `dmn.topology.edge_vertex_indices` and `cells.topology.tri_vertex_indices`, which are specific to the `Trimesh` class; but they can be adapted to other geometries.

akrivx, 2025-04-08:
That's exactly what I need, thanks.

akrivx, 2025-04-08:
Since edges are oriented such that the CCW normal points towards the outer cell, I was expecting that the vertices with indices `edge_vidx[0]`, `edge_vidx[1]`, and `outer_v` are arranged in a CCW fashion.

However, that doesn't seem to be the case, and in practice, `cross(v1 - v0, v2 - v0)` (where `v0`, `v1`, and `v2` are the vertices at the indices mentioned earlier) points in the opposite direction of `fem.normal(cells, outer_s)`.

Is there a standard way to reconstruct the inner and outer triangle normals using the edge vertices and those returned by `find_other_vertex` (instead of using `cells.topology.tri_vertex_indices`)?

gdaviet, 2025-04-08:
Oh right, sorry, the Trimesh3D version uses the opposite convention for the edge ordering compared to Trimesh2D:
```python     edge_center = 0.5 * (v1 + v0)
        edge_vec = v1 - v0
        edge_normal = wp.cross(edge_vec, tri_normal)

        # if edge normal points toward inner triangle centroid, flip indices
```

This does not matter for most use cases as the "inner" and "outer" faces are arbitrary, for for you application you'll want to flip the edge indices to build the stencil compared to the snippet above.

akrivx, 2025-04-09:
That's right, thanks.

closed 2025-05-07 by shi-eric