NVIDIA/warp #632: [BUG]

DINHQuangDung1999, 2025-04-08:
### Bug Description

Hello,
I installed Warp 1.7.0 and Isaac Sim 4.5.0. When I run the example scripts (e.g. example_rigid_force.py) and render the model into .usd, Isaac Sim could not open this file. The error description was as follow
2025-04-08 06:18:42  [Error] [omni.usd] Failed to open : /home/qdinh/Desktop/working/Warp/examples/example_rigid_force.usd
2025-04-08 06:18:42  [Error] [omni.kit.window.file.file_window] Failed to open stage /home/qdinh/Desktop/working/Warp/examples/example_rigid_force.usd: Failed to open:/home/qdinh/Desktop/working/Warp/examples/example_rigid_force.usd
Can you check if this is a bug?

### System Information

Python 3.10.12
Ubuntu 22.04.5 LTS

shi-eric, 2025-04-08:
Confirm your usd-core version and try using https://pypi.org/project/usd-core/24.11/

I expect it's the same thing as #544 and #513, a "gift" from the USD team. I heard this issue will be fixed in the future.

shi-eric, 2025-04-08:
Here's the thread in OpenUSD where they promise to be nicer https://forum.aousd.org/t/update-to-openusds-crate-usdc-versioning-support/2342

closed 2025-04-30 by shi-eric