NVIDIA/warp #636: [QUESTION] What's wrong with this kernel?

chaoming0625, 2025-04-09:
Could you help me fix the bug? I do not know what's wrong with this kernel. 

```python

import numpy as np
import warp

warp.clear_kernel_cache()

m, n, p = 1024, 1024, 0.1
spikes = np.random.random((m,)) < p
matrix = np.random.random((m, n))

TILE_SIZE = 128

def cdiv(a, b):
    return (a + b - 1) // b

@warp.kernel
def kernel(
    weights: warp.array2d(dtype=warp.float32),
    spikes: warp.array1d(dtype=warp.bool),
    out: warp.array1d(dtype=warp.float32),
):
    i = warp.tid()
    temp = warp.tile_zeros(shape=(TILE_SIZE,), dtype=warp.float32)
    for k in range(spikes.shape[0] // TILE_SIZE):
        spk = warp.tile_load(spikes, shape=(TILE_SIZE,), offset=(i * TILE_SIZE,))
        for j in range(TILE_SIZE):
            if spk[j]:
                w = warp.tile_load(weights[k * TILE_SIZE + j], shape=(TILE_SIZE,), offset=(i * TILE_SIZE,))
                temp =  temp + w
                print(w)
    warp.tile_store(out, temp, offset=(i * TILE_SIZE,))

spikes_wp = warp.array(spikes, dtype=bool)
matrix_wp = warp.array(matrix, dtype=float)
b = warp.zeros((n,), dtype=warp.float32)
warp.launch_tiled(kernel, dim=[cdiv(n, TILE_SIZE)], inputs=[matrix_wp, spikes_wp], outputs=[b], block_dim=TILE_SIZE)

```

shi-eric, 2025-04-09:
Do you plan to use autodiff on this problem? If not try setting `warp.config.enable_backward = False`. Seems like there are some bugs here we need to fix...

daedalus5, 2025-04-10:
I looked into why the backward code generation wasn't working. There are two issues:

1. `spk[j]` is a tile extract operation, and the adjoint of this operation uses atomic adds. Atomic adds won't work for booleans. So the fix there is to use integers instead.
2. `temp = temp + w` isn't supported syntax for tiles at this time. We have an update coming in version 1.8 that will support `temp += w` however.

chaoming0625, 2025-04-11:
> I looked into why the backward code generation wasn't working. There are two issues:
> 
> 1. `spk[j]` is a tile extract operation, and the adjoint of this operation uses atomic adds. Atomic adds won't work for booleans. So the fix there is to use integers instead.
> 2. `temp = temp + w` isn't supported syntax for tiles at this time (in backwards mode). We have an update coming in version 1.8 that will support `temp += w` however.

Thanks for the explanation.

daedalus5, 2025-04-11:
Quick follow-up: `temp = temp + w` isn't supported specifically in conditional expressions during adjoint generation.

closed 2025-05-01 by daedalus5