NVIDIA/warp #640: [QUESTION] How to debug missing gradients?

JonathanKuelz, 2025-04-09:
I run simulations with various meshes. Sometimes, they can take on unconventional shapes. I would like to compute gradients of some simple loss functions, e.g., the distance traveled under gravity impact in a certain time.

Usually, this works just fine (using the "hack" described in #610 to get the gradients in the first place). However, in some (unpredictable) cases, the gradients are all-zero. I have not yet found a way to reproduce this (unfortunately it is not feasible to extract standalone code to reproduce just this issue from my codebase right now), but based on my experience, this happens 
- When the meshes become too "edgy" / asymetric
- faster, when I use finer meshes (rule of thumb: at least 1200 vertices).

To demonstrate what I mean, I attached a small video of a simulation for which the gradient computation **fails**. If I change minor details about the mesh, I get nonzero gradients for all contact points again. Clearly, there is contact, and that contact must result in some non-zero impact on the final outcome. Still, the gradients are zero.

My first guess was that there are undetected `nan` or `inf` values somewhere in the computation of the gradients - but going through all computations step by step, I can't find anything that's wrong. The visualization of the tape also looks good (and does not change between "good" and "dysfunctional" simulations).

I have no idea how to detect what's going wrong here anymore - if there's another debugging tool or a hunch of where I should look, I'd appreciate it!

https://github.com/user-attachments/assets/470a4610-df46-44e5-bb94-312865040a5f

JonathanKuelz, 2025-04-11:
Thanks for the suggestion. There is only one warning, but as far as I can tell, it should not be problematic.

> Array passed to argument **contact_pairwise_counter** in kernel handle_contact_pairs ... is being written to after it has been read from within the same kernel

This counter is only used to possibly interrupt the computation, it is not used in any "physical" computation.

daedalus5, 2025-04-15:
There could be some instability in the solver gradient computations (especially with irregular meshes). I think a repro would be necessary for us to be able to say more.

JonathanKuelz, 2025-04-24:
Unfortunately, I can't provide the source code, but I think I figured out the source of the issue.

In contact-rich simulations, the gradients for the `body_q` attributes tend to explode. A simple example is a sphere rolling on a plane. Here is a small script to reproduce this and some screenshots of the position and gradients tracked:

```python
import trimesh
import wandb
import warp as wp
import warp.sim

run = wandb.init(name='debug_gradients')

mesh = trimesh.creation.uv_sphere(.5)
builder = wp.sim.ModelBuilder(up_vector=wp.vec3(0., 0., 1.))
b = builder.add_body(origin=wp.transform(wp.vec3(0., 0., 0.5), wp.quat_identity()))
builder.add_shape_mesh(b, pos=wp.vec3(0., 0., 0.), rot=wp.quat_identity(),
                          mesh=wp.sim.Mesh(vertices=mesh.vertices.tolist(), indices=mesh.faces.tolist()),
                       ke=1e3, kd=1e2, kf=1e2)
model = builder.finalize(requires_grad=True)
model.gravity = wp.vec3(2.53, 0., -9.46)
integrator = wp.sim.SemiImplicitIntegrator()

steps = 1000
states = [model.state() for _ in range(steps)]

tape = wp.Tape()
with tape:
    for step in range(steps - 1):
        s0 = states[step]
        s1 = states[step + 1]
        s0.clear_forces()
        wp.sim.collide(model, s0)
        integrator.simulate(model, s0, s1, 0.01)

s1.body_q.grad.assign(wp.transformf(1., 0., 0., 0., 0., 0., 0.))
tape.backward()

for state in states:
    q = state.body_q.numpy()
    g = state.body_q.grad.numpy()
    run.log({
        'x': q[0, 0], 'y': q[0, 1], 'z': q[0, 2],
        'grad_x': g[0, 0], 'grad_y': g[0, 1], 'grad_z': g[0, 2],
    })
```

[image]

Of course, this minimal scenario could be made less stiff, but even with "softer" contact models, eventually, I am observing huge gradients in the direction of the contact normals.

The vertex positions of the mesh do not "directly" impact the body position, but of course the position at time t changes with the mesh geometry, so it makes sense that gradients flow back to the mesh vertices. I still haven't figured out why / where the mesh gradients are zeroed, but so far, this has always gone together with exploding gradients in the `body_q` parameter. In the initial simulation I uploaded, these exploding gradients appeared "faster" when the mesh was more asymmetric / rolled worse.