NVIDIA/warp #641: [QUESTION] Evaluating fields at other triangle nodes during nodal assembly in `fem.integrate`

akrivx, 2025-04-10:
Hi,

I'm integrating over a triangle mesh using `fem.integrate` with `assembly='nodal'`. My integrand evaluates the trial field `u` at the sample point `s`, which corresponds to a specific node `i` of a triangle with nodes `(i, j, k)`.

Is there a way to evaluate the same or any other field at the other two nodes of the triangle — i.e., compute sample points corresponding to nodes `j` and `k`, so I can evaluate `u(s_j)` and `u(s_k)` from within the integrand?

This would allow me to compute expressions that involve not only `u(s)` at the current node, but also `u` at neighbouring nodes of the same element.

Thanks,
Chris

akrivx, 2025-04-10:
Would the following work?

```python
s_j = s
s_j.qp_index = j

s_k = s
s_k.qp_index = k

s_j = fem.at_node(u, s_j)
s_k = fem.at_node(u, s_k)
```

akrivx, 2025-04-11:
That's good to know, thanks.

My case involves assembling the full weak form:

$$\sum_{i, j} \langle v(s_i), H_{ij} u(s_j) \rangle$$

where $$u$$ and $$v$$ are the trial and test functions.

I was hoping that my integrand function could return something like this:

```python
wp.dot(v(s), Hi0 @ u(s0) + Hi1 @ u(s1) + ...)
```

The above evaluates the _trial_ function at various sampling points, but I understand this is impossible.

Is there an idiomatic way to perform the above computation, or should I fall back to a manual assembly approach?

gdaviet, 2025-04-11:
To express the original weak form, assuming that the `s_i` are located at nodes, the idiomatic way would be to define the bilinear form like
```
@fem.integrand
def form(s: fem.Sample, u: fem.Field, v:fem.Field, H:wp.array):
    Hij = ...
    return wp.dot(u(s), Hij * v(s))
```
then call the `integrate` routine with a ` fem.NodalQuadrature`. There is distinction between "nodal" integration and using a `NodalQuadrature`; the former assumes that u_i v_j = kronecker(i,j), which you don't want here. The latter simply means that the nodes will be picked as quadrature points, but makes no further assumption.

In the current version of warp there's not easy way to get the indices for `H` (in the dev branch I have added a `fem.node_index(u, s)` that will return that, but it's not pushed yet). However, that can be computed manually using something like
```
i = domain.tri_vertex_indices[s.element_index][s.test_dof_index[0]]
j = domain.tri_vertex_indices[s.element_index][s.test_dof_index[1]]
```

gdaviet, 2025-04-11:
(Sorry I first misread your last message)

Yes, it is not directly possible to evaluate the test and trial fields at distinct locations. 

Technically, a solution would be to do the integration with a single-point per element quadrature (`fem.RegularQuadrature(order=0)`), and  use a form like: 

```
@fem.integrand
def form(s: fem.Sample, u: fem.Field, v:fem.Field, H:wp.array):
    Hij = ...
    return wp.dot(u(s), Hij * v(s)).
```

The `form` will be evaluated for each test/trial node pair for each element, which should allow you to compute the desired sum. If using linear triangle shape functions their will be a scaling factor of tri_area/9 due to `u` and `v` being evaluated at the triangle center.

In the current version of warp there's not easy way to get the indices for H (in the dev branch I have added a fem.node_index(u, s) that will return that, but it's not pushed yet). 
However, that can be computed manually using something like:

```
i = domain.tri_vertex_indices[s.element_index][s.test_dof_index[0]]
j = domain.tri_vertex_indices[s.element_index][s.test_dof_index[1]]
```
(if you do this make sure to pass `assembly="generic"` to the integrate call)

In the end, manual assembly also looks like an attractive option!

akrivx, 2025-04-11:
That was very insightful, thank you.

Did you mean

`s.trial_dof_index[1]`

instead of

`s.test_dof_index[0]`

in

`i = domain.tri_vertex_indices[s.element_index][s.test_dof_index[0]]`?

gdaviet, 2025-04-11:
That was a typo thanks, edited! Or fore more clarity, you may use `fem.types.get_node_index_in_element(dof_index)` rather than doing the `dof_index[0]` indexing explicitly.

By the way, dof_index[1], or `get_node_coord_in_element(dof_index)`, corresponds to the active coordinate of the trial/test nodes; so instead of computing the dot product (u'Hv), you can likely extract directly the component that you need as 
`H[i, j][test_dof_index[1], trial_dof_index[1]]`.

akrivx, 2025-04-11:
That's very insightful, thank you.